MarlinFirmware/Marlin #23856: [BUG] Stepper motors make grinding sound with USB mass storage enabled

jackwilsdon, 2022-03-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When enabling USB mass storage (e.g. using the `BIGTREE_OCTOPUS_V1_USB` environment), the stepper motors in my printer make a grinding sound and do not move smoothly.

### Expected behavior

Motors would remain silent and move smoothly.

### Actual behavior

Motors made a grinding noise and moved jerkily.

### Steps to Reproduce

1. Build Marlin with USB mass stroage enabled
2. Note that motors do not move smoothly and produce a grinding noise

### Version of Marlin Firmware

Marlin 2.0.9.3 (Mar  5 2022 12:33:18)

### Electronics

BigTreeTech Octopus

### Add-ons

TMC2208 in UART mode

### Additional information & file uploads

I assume this is caused by some step timing inconsistencies with mass storage enabled but I'm having issues probing it at the moment.

ellensp, 2022-03-05:
Default/provided configuration is for ramps with A4988's

jackwilsdon, 2022-03-05:
Oops! I'll update the post (not sure why I thought it was a completely stock configuration 🤦).

This issue still happens even if I'm using the TMC2208s outside of UART mode (both with the `A4988` and `TMC2208_STANDALONE` driver types), so the new configuration I've attached still uses the A4988 option.

The issue also immediately goes away if I unplug the USB connection.

ellensp, 2022-03-05:
This may be related. Someone just reported in Discord "If a large number of serial port data is received, the STEP signal is delayed."  USB being chatty might qualify. they where talking about stm32f407.

jackwilsdon, 2022-03-05:
I've not been able to reproduce that myself with USB mass storage disabled (just tried sending a large amount of `M503`s over serial as fast as possible), but I've noticed that the heartbeat LED slows down dramatically (like >50%) when using USB mass storage.

BreeOge, 2022-03-05:
I have had this exact same issue described and I am using an Octopus v1.1 with TCM2209.  I have had it shut down due to error.  I have not been able to reproduce it on a consistent basis, If I am, I will post the needed steps I find.

jackwilsdon, 2022-05-05:
This issue is still present.

closed 2022-07-15 by github-actions[bot]