MarlinFirmware/Marlin #23858: [FR] (Passive nozzle lift by moving to position)

AndreasOpgenoorth, 2022-03-06:
### Is your feature request related to a problem? Please describe.

I use a dual drive system. I added a nozzle lift mechanism, which is able to move one of the two nozzles up or down by moving to a certain position. This enables the lower nozzle to be used for printing, while the other nozzle is out of the way. (Similar to the Ultimaker system)

Here is a video of this:

https://youtu.be/A0d3_HDL-Dw

There does not seem to be a possible integration of such a system in Marlin yet (MAGNETIC_PARKING_EXTRUDER is for a different system and EVENT_GCODE_AFTER_TOOLCHANGE does not seem to allow different codes for each individual tool index), but I do believe, that this might be interessting for more people in general.

### Describe the feature you want

Either function similar to MAGNETIC_PARKING_EXTRUDER or an further adaptation for EVENT_GCODE_AFTER_TOOLCHANGE.

You should be able to add to each indiviual T0, T1, ... certain gcodes, which will be executed on triggering the nozzle switch.

For me this would be:

T0 triggers G1 X340 F5000
T1 triggers G1 X0 F5000

AndreasOpgenoorth, 2022-03-06:
Also, if anyone knew some workaround to enable this system easily, I would be happy to hear about it.

DerAndere1, 2022-03-10:
I think this can be closed. Have you had a look at current Marlin? I added options `EVENT_GCODE_AFTER_TOOLCHANGE_T0` and `EVENT_GCODE_AFTER_TOOLCHANGE_T1` some time ago, although it was not yet added to the documentation. If it does not work, additionally enable `EVENT_GCODE_AFTER_TOOLCHANGE_ALWAYS_RUN`

AndreasOpgenoorth, 2022-03-10:
Ok, thank you. Can be closed then.

closed 2022-03-10 by AndreasOpgenoorth