MarlinFirmware/Marlin #23859: [BUG] Delta Calibration Does Not Work

fusiondew, 2022-03-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

There are many things within the Tower Angle Trim and Diagonal Rod Trim calculations that seem to be wildly inaccurate, on top of having absolutely no documentation, AND nobody knowing how any of it actually works.

1) M665 Rod Trim exists. But that's about it. There is no documentation whatsoever on how to accurately calibrate these values or how these values are used. They are simply "there for you to look at". The only source for rod calibration is a thingiverse object/post and the post seems to use old variables from marlin 1.9 that work differently than the m665 Rod Trim values. The people that have managed to calibrate these values via trial and error for dimensionally accurate prints have values showing that some rods are 10mm longer than others, when in reality the accuracy of their rods are within 1mm. My rods are within 0.12mm accuracy and I am having to adjust by 5mm for one of them to come close. I have personally spent 2 months working on dimensional accuracy in marlin alone and haven't achieved it yet. Lots of posts in different forums, lots of headaches, lots of trial and error, lots of going through the code, yet nobody knows how this works in marlin 2.0.

2) Delta_Tower_Angle_Trim states "get these values from G33 Auto Calibrate", yet the G33 values for this are vastly inaccurate with tower angles off by 1-2 degrees for many people.

I'm not sure if Marlin has completely dropped further development for Delta printers, and if that is the case I retract my statements and will look at other options. It seems at this point Marlin is not a viable option for Delta printers.

### Expected behavior

I expect there to be documentation for something as important as individual rod length and tower angle trim. I also expect the calculations to work properly.

### Actual behavior

There is no documentation anywhere, and nobody in 6+ public forums that knows how it works. Values needed to adjust these variables are arbitrary and can only be found with trial and error. Once the values are found, they do not reflect the actual build of the machine in any way.

### Steps to Reproduce

Try to calibrate individual rod lengths and tower angle trim in marlin 2.0.
I dare you.

### Version of Marlin Firmware

Marlin 2.0.9.3

### Printer model

Custom Precision Machined All Metal Rostock Printer

### Electronics

BTT SKR2 STM32F429, TMC2209 drivers, Conductive Nozzle and/or Piezoelectric Probe, 1.8A NEMA 17 steppers. 

### Your Slicer

Slic3r

### Host Software

Repetier Host

sjasonsmith, 2022-04-05:
> Try to calibrate individual rod lengths and tower angle trim in marlin 2.0.
> I dare you.

Please provide exact instructions to reproduce the issue.

drf5n, 2022-04-12:
Is this documentation at https://marlinfw.org/docs/gcode/M665.html in error?

[image]

In particular, why are there two `B` words, one for `Beta (Tower 2) diagonal rod trim` and one for `Delta calibration radius`?  

It doesn't match the source:
 https://github.com/MarlinFirmware/Marlin/blob/2893060302b7251cb6d5cd258f264b1a87eedee4/Marlin/src/gcode/calibrate/M665.cpp#L35-L62

thisiskeithb, 2022-04-12:
> Is this documentation at https://marlinfw.org/docs/gcode/M665.html in error?

Thanks for the heads up! https://github.com/MarlinFirmware/MarlinDocumentation/pull/445

sjasonsmith, 2022-05-30:
Unfortunately probably the only way to see any improvement in this part of Marlin is to become an expert yourself and seek to understand/fix issues.

I haven't tested delta calibration for a while, but when I did use it it appeared to work well enough. Did it actually match my geometry? I don't know. Whatever it did was good enough to let me print on the bed without using bed leveling, which satisfied my needs.

With the number of variables involved in delta calibration I'm not surprised that a combination of settings that don't actually represent the actual hardware could end up providing decent results. To really verify the accuracy of the calibration routines I think you would need to do it using mathematical simulations, since actually building physical machines to that degree of known precision would be pretty difficult.

LVD-AC, 2022-09-02:
> In particular, why are there two `B` words

Old B "calibration radius" was abandonned and should no longer be there.

> yet the G33 values for this are vastly inaccurate with tower angles off by 1-2 degrees for many people

Changing the build of the triangles (losening the crews and tapping with a hammer on the sides that are reprted to big) does change the tower angle correction calculated by G33 accordingly; so there is a physical relation. G33 however only calibrates so the printhead moves in a plane parallel to the build surface.

This of course when probing can be done with sufficient accuracy : see #22853. But nobody seems to bother to look into what went wrong between 2.0.7 and 2.0.8. As long as probing a certain point, moving the effector around a bit and probing the same exact point again gives different and diverging results one can forget to calibrate a delta.

> M665 Rod Trim exists. ... They are simply "there for you to look at". ... showing that some rods are 10mm longer than others,

Trying to change rod lenghts to unrealistic values in order to correct for dimentional inaccuracy isn't the right way of doing. It just creates a set of different problems : changing rod lenghts, also should be reflected in a changed delta radius along that axis and it end up as straight lines being printed as arcs. So rod trim without corresponding delta radius trim in M665 is IMO only "there for you to look at", but has no real use. It also changes the perceived tower angles and disturbs the calibration of the flat printing plane. Recalibrating the printing plane by changing the tower angles to unrealistic values as well, undoes the dimentional effects of the rod lenght changes. Hense it diverges to completely unrealistic values of all the parameters.

Unless someone explains to me how dimentinal calibration can be done correctly, only then I'm willing to give it a try to incorperate it in G33.

pyzahl, 2024-01-12:
Any updates here? I just ran into those issues after updating my very of 2019 Delta's hardware to BTT SKR MINI E3 V3  + TFT35 and after getting all to work I an stuck here :(  The 2019 Marlin version worked just fine and I was only expecting to to get better.... not this :(

I have even worse miss behaving to near crashing fast into my switch after a initial slow and correct gentle Z center bed probe and keeps doing this in the center oddly.

Homing and XYZ motion manually do work all fine. So the Z-Probe.

closed 2024-09-19 by github-actions[bot]