MarlinFirmware/Marlin #23860: [BUG] All icons missing on Ender 3v2

Winterfoxxy-0, 2022-03-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling Marlin for my ender 3v2 (on the default branch, bugfix won't compile), it works fine, however there are no icons on the screen. For whatever reason, my display refuses to flash a DWIN_SET, so that's out of the question. I'll attach some images, maybe they'll be of assistance.
[image]
[image]

### Bug Timeline

new bug, first time building

### Expected behavior

All icons exist on the display.

### Actual behavior

All icons missing.

### Steps to Reproduce

1. Build Marlin for the ender 3v2 on the default branch (2.0.x)
2. Flash it like normal

### Version of Marlin Firmware

2.0.93

### Printer model

creality ender 3 v2, main board 4.2.2

### Electronics

Stock

### Add-ons

CRTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

Winterfoxxy-0, 2022-03-06:
Here are the config files:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/8193358/config.zip)

Winterfoxxy-0, 2022-03-06:
Also it appears that some text is offset in a strange way, and some text is highlighted when it shouldn't be.

thisiskeithb, 2022-03-06:
> my display refuses to flash a DWIN_SET, so that's out of the question

You need to figure out why because that's why you're having this issue. Try [following the readme](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3%20V2) again. Try some other name-brand SD cards as well.

closed 2022-03-06 by thisiskeithb

mriscoc, 2022-03-07:
Maybe you have a DACAI screen unit, you can check which screen you have by comparing it with the pictures from here: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-update-the-display

Winterfoxxy-0, 2022-03-07:
i have to go to school so i will in the afternoon