MarlinFirmware/Marlin #23861: [FR] (feature summary)[FR] Z Touchoff Probe for Multiple Nozzles

KillerBug, 2022-03-07:
### Describe the feature you want

I designed a printer and built a prototype.  It's based on a Voron 2.4 but it is very much modified.  It has what I believe to be a new invention in it, and I open source'd it.  Problem is I am having no luck getting started on the bit of code that will make it work.

There is a switch in my printer that the nozzles can move to (it's at the front of the bed).  The idea is that it will be level with the bed surface, or at least pretty close with a firmware offset.  Using this switch, the offset for the inductive probe should be able to be calculated.  It should also detect level differences between the nozzles if there are any, ideally with some kind of menu to say something like "move the left nozzle down 0.5mm and try again".  The board used is an Octopus 1.1 (on Marlin of course).

Here's (what I believe to be) the first printer in the world with this hardware: https://grabcad.com/library/not-a-voron-2-5-r103-1

dougabugg, 2022-03-07:
I've subscribed to this issue. I'm a software developer, and talked a little bit to killerbug about their printer and am hoping to dive into and understand the Marlin Firmware project eventually, but haven't had enough time too yet. Would if be useful to share the config file the printer is currently using?
Thanks!

InsanityAutomation, 2022-03-09:
Can you describe more how the process you have in mind would deviate from the existing G425 routines?

KillerBug, 2022-03-10:
> Can you describe more how the process you have in mind would deviate from the existing G425 routines?

A couple of things different from G425.  First there is no cube/bolt/etc to crash into; the button top is level with the bed.  Other big difference is the process, this is tied in with bed leveling.  Maybe a step-by-step would help:
01.) Machine homes x & y & all 4 z motors
02.) Machine does whatever z motor offsets there may be
03.) Machine does automatic bed leveling with an inductive probe, although it is incomplete because the distance from the bed to where the inductive probe triggers isn't known yet.
04.) Machine touches off with first nozzle.  The order here matters; this should happen after the bed leveling because that will change the location of the switch that is mounted to the bed.
05.) Machine touches of with second nozzle.
06.) First and second nozzle touch offs are compared and (if relevant) adjustment instructions go on screen.  If they are within a "close enough" range then it takes the average and uses that to complete the bed leveling.
07a.) A few bonus features that people might get use of but that I don't have hardware for yet:
07b.) Ability to use an actuator (solenoid probably) to adjust the height of one of the nozzles.
07c.) Ability to use with more than two nozzles (I think E3D still makes a hotend with 4 nozzles).
07d.) Ability to use it with a single nozzle (no nozzle height work is needed but the post-leveling touch-off for probe calibration might still be useful.)

InsanityAutomation, 2022-03-10:
On the process above, if you are calculating the deviation at a fixed point, I dont see how bed leveling before or after is relevant. Taking the measurement difference in the same state is the critical part. It may be important to follow G34, but G29 should be invisible to this process. The mesh application should be shifted when tool is changed firstoff so position calculated to shift should match, and second we typically disable leveling transforms for calibration routines like this.

Lulzbot toolheads can adjust servo positions for the mentioned height adjustment, so that may be a viable reference point since its an open source design you can replicate.

In all honesty, Im not sure you couldnt follow the entire G425 procedure with this. You should get the X/Y positions from the rising edge of the sensor provided the Z position used is within the potential deviation so its always within sensing range as it swings over.

KillerBug, 2022-03-10:
The point isn't fixed; it's attached to the bed, so it will be in a slightly different position after bed leveling.  The offset read between the nozzles and the inductive probe will also change with the pre-heating.

Nice to know about the Lulzbot toolhead; didn't know that existed.  Very cool...and expensive!

I'm having a hard time trying to see how G425 could do the job...does it work with buttons like this or do I have to use a conductive cube/bolt/etc like in the description?

DerAndere1, 2025-09-07:
G425 does automatic calibration of hotend offsets. You can either use a conductive nozzle+block or you can attach a tool-setter to a pin and define that pin as CALIBRATION_PIN. The tool setter can be a mechanical switch or an optical device or a similar sensor.