MarlinFirmware/Marlin #23866: [FR] Z fade for Linear Bed Levelling

blex-max, 2022-03-07:
### Is your feature request related to a problem? Please describe.

I'm not sure why z fade height is unavailable for linear bed levelling

### Are you looking for hardware support?

No

### Describe the feature you want

I would like to be able to fade the Z correction when using linear levelling, which I find to be most reliable.

### Additional context

Apologies for the brevity, I can't think of much else to add.

tombrazier, 2022-03-07:
Linear bed leveling rotates the entire model to sit squarely on the bed. There is no need for z fade - indeed it would introduce distortions in the model.

blex-max, 2022-03-07:
I did not realise this, could you point me towards the documentation? Why does z fade not then introduce distortions in mesh levelling - it seems to follow to me that as you fade it out you would get distortions, but people anecdotally report models are in fact more warped if you don't fade. Might this not be true of linear also?

Thanks!

tombrazier, 2022-03-07:
There is some info in the G29 and bed leveling documentation. https://marlinfw.org/docs/gcode/G029.html and https://marlinfw.org/docs/features/auto_bed_leveling.html. Apart from that, I have been asking a lot of questions lately and working on the code - which is definitely not always the easiest way to answers!

Z fade does introduce distortions. At best, the side of the model facing the bed is warped a bit whilst the rest of the model is not. At worst (and I did this recently accidentally) your home zero might be some way from the bed and then z fade seriously increases or decreases your model height. You can only avoid the former by getting a flatter bed. The latter can be avoided by setting `MANUAL_Z_HOME_POS` so that home zero is at average bed height. (Which I have now done for my machine!) I am pondering submitting a pull request in which BILINEAR does this automatically.

Incidentally why use LINEAR and not BILINEAR? I understand from a recent conversation that LINEAR was one of the earlier ways of doing leveling.

InsanityAutomation, 2022-03-07:
The biggest risk with fade out height is when people use it on a bed that is simply tilted not actually warped and they create a leaning tower of Pisa... But in the best case as mentioned above I agree that it will force all of the distortion to the bottom surface of the model and the rest of it would come out smooth. Personally it's a feature that I stay far away from as even when it's working correctly you will have uneven extrusion throughout the layers with adjusted compensation. This could create potential structural weaknesses in the part.

Roxy-3D, 2022-03-23:
If you use Z-Fade and care about not having any distortion....    You need to understand what kind of distortion you want to avoid.

For example...  suppose 1/2 of your mesh points are below 0.00 mm and half are above 0.00 mm.   In that case, the total height of the part will be what ever your lowest mesh point is (below 0.00 mm) plus the height of the part.    The reason is the nozzle will dip to trace the bed on the first layer and go below 0.00 mm to get adhesion.   But at the top of the part everything will be tracking smoothly on a flat plane.

Alternatively...    if your lowest points of the mesh are at 0.00 mm and any distortion that is being corrected is a positive value for the mesh point, your part will end up exactly the right height.   It will go from 0.00 mm to what ever height it is at.   But...  you will have a little extra density of material (that very well could cause a distortion) where the nozzle had to raise to accommodate a mesh point correction.

PS.   The Z-Fade height correction first showed up in 3D Printing in the UBL bed leveling system.

thinkyhead, 2022-04-10:
Using Z fade for an object that is initially printing at an angle will result in the area below the fade height having a slight curve as it goes from printing the object at an angle to printing the object true to the gantry / bed. The result is not desirable.

closed 2022-04-10 by thinkyhead