MarlinFirmware/Marlin #23877: [BUG] Ender 3 v2 startup progress bar

steljwagh, 2022-03-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When `#define DWIN_CREALITY_LCD_JYERSUI ` the Ender 3 v2 displays a weird second progress bar after startup.

I also tried `#define DWIN_CREALITY_LCD_ENHANCED`, then there is a screen reset issue which also causes a weird artefact after starting up, which causes every item to disappear and reappear on dialling the rotary encoder don't intend this setting but maybe it's relevant.

It doesnt seem to affect the printer's working.

I compile with the STM32F103RE_Creality(512k) environment for a BOARD_CREALITY_V427 motherboard.
During compiling I got this error:

```
Marlin\src\module\settings.cpp: In static member function 'static void MarlinSettings::reset()':
 3206 |   FSTR_P const hdsl = F("Hardcoded Default Settings Loaded");
      |                ^~~~

```
Doesnt seem to relate to the display issue, but every information could be of help :)

My DWIN_SET has been updated to the files included from the marlin configuration example repo.

https://user-images.githubusercontent.com/20403141/157552088-39781ca9-d9ea-4097-8d91-ec87635a16cc.mp4

 

### Bug Timeline

Unknown, didn't update for a year

### Expected behavior

I didn't expected to see a progress bar on top of menu items while nothing was selected or triggered.

### Actual behavior

A progress bar on top of menu items right after the startup

### Steps to Reproduce

It happens every bootup

### Version of Marlin Firmware

burgfix 2.0.x

### Printer model

Ender 3 v2

### Electronics

4.2.7 motherboard

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

ellensp, 2022-03-10:
Do you have the sdcard inserted?  Try it without sd card inserted. then insert the sdcard after boot, do you get a new progress bar on insertion of the sdcard?

steljwagh, 2022-03-10:
No, I don't have the sdcard inserted. When booting up the second progress bar is visible with or without the sd card inserted. When i insert the sd card after boot i do get a new progress bar on insertion of the sdcard.

ellensp, 2022-03-10:
That points me at   `#define REINIT_NOISY_LCD 1  // Have the LCD re-init on SD inserting`

Could you try a little code change.

In Marlin/src/inc/Conditionals_post.h is
```
#if PIN_EXISTS(SD_DETECT) && NONE(HAS_GRAPHICAL_TFT, LCD_USE_DMA_FSMC, HAS_FSMC_GRAPHICAL_TFT, HAS_SPI_GRAPHICAL_TFT, IS_DWIN_MARLINUI, EXTENSIBLE_UI)
  #define REINIT_NOISY_LCD 1  // Have the LCD re-init on SD insertion
#endif
```
change this to include your DWIN_CREALITY_LCD_JYERSUI ie chnage to the following
```
#if PIN_EXISTS(SD_DETECT) && NONE(HAS_GRAPHICAL_TFT, LCD_USE_DMA_FSMC, HAS_FSMC_GRAPHICAL_TFT, HAS_SPI_GRAPHICAL_TFT, IS_DWIN_MARLINUI, EXTENSIBLE_UI, HAS_DWIN_E3V2)
  #define REINIT_NOISY_LCD 1  // Have the LCD re-init on SD insertion
#endif
```

steljwagh, 2022-03-10:
Wow, you fixed both issues.

I tried the DWIN_CREALITY_LCD_JYERSUI and the progress-bar after bootup is now gone and the display behavior is exactly as expected, also upon removal/insertion of sdcard.

I took the opportunity to also test DWIN_CREALITY_LCD_ENHANCED and the weird display glitch after bootup is also gone now.

Thank you for this fix! :)

ellensp, 2022-03-10:
bet me to it by 5 minutes :) 
Closing this issue as it is solved

closed 2022-03-10 by ellensp