MarlinFirmware/Marlin #23881: [BUG] M114 D zeros out current position ...

softfoot, 2022-03-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

While debugging BILINEAR leveling via "manual controls" on Repetier-Host I could move the head around OK and the display on Repetier-Host tracked it OK, but when I issued M114 D to see what the leveling status was like it zeroed out the x and y positions on Repetier-Host and if I attempted a further move in X & Y it got it wrong so I suspect the problem is in Marlin rather than RH.
It's a bit of a pain because while exploring what leveling is doing at various bed positions I have to do a home after every M114 D to get meaningfull results.
Running v2.0.9-3 on a Geeeteck i3 pro C -- Repetier-Host is the latest.
Dave

### Bug Timeline

I suspect its been there a while

### Expected behavior

I should be able to do the sequence ...
G28, M114 D, MOVE X/Y, M114 D, MOVE X/Y, M114 D

### Actual behavior

The x & y axes became meaningless.

### Steps to Reproduce

G28
M114 D
MOVE X/Y
M114 D

### Version of Marlin Firmware

2.0.9-3

### Printer model

Geeeteck i3 pro C 

### Electronics

Standard controller

### Bed Leveling

ABL Bilinear mesh

### Host Software

Repetier Host

descipher, 2022-03-10:
Nope, works correctly on Repetier-Host V2.2.3

ellensp, 2022-03-11:
example configs for this machine says the motherboard is a BOARD_GT2560_REV_A, so this is just atmega2560

Testing on a ramps

```
M114 D
> X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
> 
> Logical: X: 0.000 Y: 0.000 Z: 0.000
> Raw:     X: 0.000 Y: 0.000 Z: 0.000
> Stepper: X:0 Y:0 Z:0 E:0
> FromStp: X: 0.000 Y: 0.000 Z: 0.000 E: 0.000
> Diff:    X: 0.000 Y: 0.000 Z: 0.000 E: 0.000
> ok
G1 X20Y20
> ok
M114 D
> X:20.00 Y:20.00 Z:0.00 E:0.00 Count X:1600 Y:1600 Z:0
> 
> Logical: X: 20.000 Y: 20.000 Z: 0.000
> Raw:     X: 20.000 Y: 20.000 Z: 0.000
> Stepper: X:1600 Y:1600 Z:0 E:1
> FromStp: X: 20.000 Y: 20.000 Z: 0.000 E: 0.002
> Diff:    X: 0.000 Y: 0.000 Z: 0.000 E: 0.002
> ok
G1 X40Y40 
> ok
M114 D
> X:40.00 Y:40.00 Z:0.00 E:0.00 Count X:3200 Y:3200 Z:0
> 
> Logical: X: 40.000 Y: 40.000 Z: 0.000
> Raw:     X: 40.000 Y: 40.000 Z: 0.000
> Stepper: X:3200 Y:3200 Z:0 E:1
> FromStp: X: 40.000 Y: 40.000 Z: 0.000 E: 0.002
> Diff:    X: 0.000 Y: 0.000 Z: 0.000 E: 0.002
> ok
```

This issue is most likely host related. not marlin related.

Test this with a serial console, Iike I did.

closed 2022-03-11 by ellensp