MarlinFirmware/Marlin #23882: [BUG] echoed data reflected in 'unknown command' errors

letrow, 2022-03-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

https://github.com/MarlinFirmware/Marlin/issues/2890 reported an strange phenomenon wherein data coming from Marlin somehow got mixed with transmitted g-code commands, resulting is 'unknown command' errors. This was in 2016 and the issue was closed (without a clear solution). Now I am developing a simple host interface ('marlinterface'  - see [githup](https://github.com/letrow/3D-O-Matic-GUI) and I ran into the same problem. This happened on Marlin 1.0.0 running on a RAMPS 1.4 board so I re-tested on Marlin 2.0.9.2  on a different RAMPS 1.4 board and there the issue turns up even worse. Although the old issue 2890 was closed because someone claimed a bug in the host software, I am under the impression that there is a flaw in Marlin's serial data handling under heavy traffic. I will continue testing and come back here when there is new information. Host: Linux 5.4.0

### Bug Timeline

Ol dissue, started in 2016, issue 2890 - see above

### Expected behavior

I expect incoming data to be completely separated from transmitted data

### Actual behavior

Transmitted data sometime contains fragments of incoming data resulting in 'unknown command' errors.

### Steps to Reproduce

1. send M155 S3 to start periodic position reports or disable cold extrusion and set extruder temp low so many cold extrusion prevented messages are sent by Marlin.
2. start sending g-code commands taking care that not more than (say) 100 bytes at a time are transmitted before the 'ok'-ack has been received See my transport.py module in https://github.com/letrow/3D-O-Matic-GUI for details on how I did that.
3. monitor the replies from Marlin and observe that sooner or later the 'unknown command' error occurs. This can take a couple of minutes but will eventually happen. Also, I noted that changing the buffer size into a different value has no effect. However assuming too large a buffer in Marlin causes g-codes to get lost.

### Version of Marlin Firmware

1.0.0 and 2.0.9.2

### Printer model

Prusa i3 or no printer (stand-alone RAMPS board only)

### Electronics

Stock RAMPS 1.4

### Your Slicer

Slic3r

### Host Software

Other (explain below)

ellensp, 2022-03-10:
You cannot have two threads running on the same port at the same time. they interfere with each others

ellensp, 2022-03-11:
This is not a marlin issue, it is a issue with your Host software

closed 2022-03-11 by ellensp