MarlinFirmware/Marlin #23889: [BUG] error: 'MYSERIAL' was not declared in this scope; did you mean 'MYSERIAL2'?

tiagofreire-pt, 2022-03-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The current bugfix Marlin code generates an error while compiling:

```
Marlin/src/HAL/STM32/../../inc/../core/serial.h:72:26: error: 'MYSERIAL' was not declared in this scope; did you mean 'MYSERIAL2'?
   72 |   #define _SERIAL_LEAF_1 MYSERIAL
      |          
```                ^~~~~~~~
      
      
      

### Bug Timeline

New

### Expected behavior

Show compile without errors.

### Actual behavior

Compile with errors.

### Steps to Reproduce

Compiling with the following .h configuration files.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Ender 3 V2

### Electronics

BTT SKR E3 MINI V3

### Add-ons

Creality Ender 3 V2 LCD with proper custom made cable

### Additional information & file uploads

configuration.h: https://pastebin.com/raw/QyGwbg0E

configuration_adv.h: https://pastebin.com/raw/QhBR8pUe

tiagofreire-pt, 2022-03-11:
Additional errors:

```
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:56,
                 from Marlin/src/HAL/STM32/MarlinSerial.cpp:27:
Marlin/src/HAL/STM32/../../inc/../core/serial.h:127:56: error: template argument 1 is invalid
  127 |   typedef MultiSerial< REPEAT_1(NUM_SERIAL, _S_MULTI) 0> SerialOutputT;
      |                                                        ^
Marlin/src/HAL/STM32/../../inc/../core/serial.h: In function 'void SERIAL_CHAR(char)':
Marlin/src/HAL/STM32/../../inc/../core/serial.h:150:47: error: request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
  150 | inline void SERIAL_CHAR(char a) { SERIAL_IMPL.write(a); }
      |                                               ^~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio/build/STM32G0B1RE_btt/src/src/HAL/STM32/MarlinSerial.cpp.o] Error 1

```

thisiskeithb, 2022-03-11:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22299

closed 2022-03-11 by thisiskeithb