MarlinFirmware/Marlin #23892: [BUG] FW retraction adds extra recover length to both extruders

GameFan01, 2022-03-12:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

The FWRETRACT function adds the extra length on recovery to all mixing extruders (even if only one is selected and active) causing the secondary filament to slowly be added to the primary, in my case causing the print to be grey instead of white.

### Bug Timeline

Bug is tied to enabling the FWRETRACT function

### Expected behavior

The expected behavior is for all extruders to retract and then recover while adding extra recover length only to the active extruder.

### Actual behavior

All extruders retract simultaneously and recover simultaneously while adding the extra recover lenght to all extruders even if some are not supposed to be used.

### Steps to Reproduce

Enable the FWRETRACT function in firmware and set the Unretr. mm in the printer menu to anything other than 0.

### Version of Marlin Firmware

Marlin 2.0.9.1

### Printer model

Creality ender 3

### Electronics

MKS GEN L V1.0 and TMC2208 drivers used in legacy mode

### Add-ons

Reprap 12864 full graphic LCD

### Bed Leveling

No Bed Leveling

### Your Slicer

Prusa Slicer

### Additional information & file uploads

[Configuration - Copy.zip](https://github.com/MarlinFirmware/Marlin/files/8237963/Configuration.-.Copy.zip)

zvonka, 2022-04-09:
I had the same issue on an SKR 1.4 with 2209 drivers about a year ago I built for my significant other. 

I was able to barel kludge together a fix by more or less screwing around with my slicer's gcode settings (cura, or simplify3d, i cant remmeber which) to kind of mostly get it workign by forcing it to retract after every tool change the extra distance before beginning the next loop, the extra filament that was fed as  ar result of this bug.

The prints looked ok but definitely you could still clearly tell that it wasn't quite right because of all the blobs this created

GameFan01, 2022-04-09:
Thank you for the workaround. I haven't thought about the after tool change gcode. 
Shame there aren't any resources about the marlin coding, otherwise I'd try to modify the firmware myself..

closed 2022-06-19 by github-actions[bot]