MarlinFirmware/Marlin #23893: Dual extruder for singlenozzle ==>DWIN_LCD_PROUI

LoloMake60, 2022-03-12:
### Is your feature request related to a problem? Please describe.

Dual extruder for singlenozzle ==>DWIN_LCD_PROUI 
Not E1 on display. Why?

### Describe the feature you want

Is it possible to add the second extruder on the display DWIN_LCD_PROUI ?
- Change filament E0
- Change filament E1
- Move E0
- Move E1
- .....
Thanks.

ellensp, 2022-03-16:
@mriscoc for your consideration.

mriscoc, 2022-03-16:
Currently DWIN_LCD_PROUI only supports a single extruder. Maybe MarlinUI can handle multiple extruders on the DWIN color display.

thinkyhead, 2022-04-26:
> MarlinUI can handle multiple extruders

Indeed it can! and other UI implementations should do their best to follow the patterns established in `MarlinUI`, or simply adopt `MarlinUI` for their menu systems –so they gain all features– while still fancying up the Info Screen according to stylistic preferences. New screens can be backported to `MarlinUI` for maximum communal benefit.