MarlinFirmware/Marlin #23894: [BUG] TMC2208 UART on ENDER 3V2 compilation error

stikzoo, 2022-03-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I can't compile a firmware, I find an issue with this, so I deleted .pio folder then retry to compile but it don't work.

### Expected behavior

I would like to compile a firmawre inside visual code with PlatformIO

### Actual behavior

> Executing task: C:\Users\h95vi\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RET6_creality (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <TMCStepper> 0.7.3
|   |-- <SoftwareSerialM> 1.0.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\VirtIOSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\abi.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\board.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\main.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\new.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\pins_arduino.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\stm32\OpenAMP\libmetal\device.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\stm32\OpenAMP\libmetal\generic\condition.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\stm32\OpenAMP\libmetal\generic\cortexm\sys.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_device.c.o
[... 274 lines cut ...]
### Electronics

Creality 4.2.2

### Add-ons

CR touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Additional information & file uploads

[Configuration_adv_.txt](https://github.com/MarlinFirmware/Marlin/files/8238082/Configuration_adv_.txt)
[Configuration_.txt](https://github.com/MarlinFirmware/Marlin/files/8238083/Configuration_.txt)
[pins_CREALITY_V4_.txt](https://github.com/MarlinFirmware/Marlin/files/8238084/pins_CREALITY_V4_.txt)

ellensp, 2022-03-13:
Provided configs upgrade to bug fix compiles without issues 
[Configuration_bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/8239115/Configuration_bugfix.zip)

ellensp, 2022-03-13:
Also build_flags = (add) -DHAVE_SW_SERIAL is not required.  loading TMCStepper will load it 
```
Dependency Graph
|-- <TMCStepper> 0.7.3
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
```

stikzoo, 2022-03-13:
Thank you, I tried it. SoftwareSerialM in library PlatFormIO is not required?
I have another issue on 3 GCODE command for TMC stepper and on feedrate.
> Executing task: C:\Users\h95vi\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RET6_creality (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing TMCStepper @ ~0.7.3

Unpacking  [------------------------------------]    0%
Unpacking  [------------------------------------]    1%
Unpacking  [#-----------------------------------]    3%
Unpacking  [#-----------------------------------]    4%
Unpacking  [##----------------------------------]    6%
Unpacking  [##----------------------------------]    8%
Unpacking  [###---------------------------------]    9%
Unpacking  [####--------------------------------]   11%
Unpacking  [####--------------------------------]   13%
Unpacking  [#####-------------------------------]   14%
Unpacking  [#####-------------------------------]   16%
Unpacking  [######------------------------------]   18%
Unpacking  [#######-----------------------------]   19%
Unpacking  [#######-----------------------------]   21%
Unpacking  [########----------------------------]   22%
Unpacking  [########----------------------------]   24%
Unpacking  [#########---------------------------]   26%
Unpacking  [##########--------------------------]   27%
Unpacking  [##########--------------------------]   29%
Unpacking  [###########-------------------------]   31%
Unpacking  [###########-------------------------]   32%
Unpacking  [############------------------------]   34%
Unpacking  [############------------------------]   36%
Unpacking  [#############-----------------------]   37%
Unpacking  [##############----------------------]   39%
Unpacking  [##############----------------------]   40%
Unpacking  [###############---------------------]   42%
Unpacking  [###############---------------------]   44%
Unpacking  [################--------------------]   45%
Unpacking  [#################-------------------]   47%
Unpacking  [#################-------------------]   49%
Unpacking  [##################------------------]   50%
Unpacking  [##################------------------]   52%
Unpacking  [###################-----------------]   54%
Unpacking  [####################----------------]   55%
Unpacking  [####################----------------]   57%
Unpacking  [#####################---------------]   59%
Unpacking  [#####################---------------]   60%
Unpacking  [######################--------------]   62%
Unpacking  [#######################-------------]   63%
Unpacking  [#######################-------------]   65%
Unpacking  [########################------------]   67%
Unpacking  [########################------------]   68%
Unpacking  [#########################-----------]   70%
Unpacking  [#########################-----------]   72%
Unpacking  [##########################----------]   73%
Unpacking  [###########################---------]   75%
Unpacking  [###########################---------]   77%
Unpacking  [############################--------]   78%
Unpacking  [############################--------]   80%
[... 406 lines cut ...]
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\e3v2\marlinui\ui_status_480x272.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\extui\ui_api.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\fontutils.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\lcdprint.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\marlinui.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\menu\menu.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\menu\menu_advanced.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\menu\menu_configuration.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\menu\menu_game.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\menu\menu_job_recovery.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\menu\menu_main.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\menu\menu_media.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\menu\menu_mixer.cpp.o
Marlin\src\lcd\marlinui.cpp: In static member function 'static void ManualMove::task()':
Marlin\src\lcd\marlinui.cpp:774:60: error: 'manual_feedrate_mm_s' was not declared in this scope; did you mean 'feedrate_mm_s'?
  774 |         const feedRate_t fr_mm_s = (axis < LOGICAL_AXES) ? manual_feedrate_mm_s[axis] : XY_PROBE_FEEDRATE_MM_S;
      |                                                            ^~~~~~~~~~~~~~~~~~~~
[... cut ...]
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\marlinui.cpp.o] Error 1
=============================================================== [FAILED] Took 59.07 seconds ===============================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:59.070
========================================================== 1 failed, 0 succeeded in 00:00:59.070 ==========================================================

ellensp, 2022-03-13:
That is a user error
Using miss matched configs files and code
See this for details
https://github.com/MarlinFirmware/Marlin/issues/23656#issuecomment-1025441678

stikzoo, 2022-03-13:
Thanks a lot I will check this.

stikzoo, 2022-03-13:
Ok it is done for the feedrate problem but this is unsolved:
Marlin\src\gcode\feature\trinamic\M569.cpp: In function 'void set_stealth_status(bool, int8_t)':
Marlin\src\gcode\feature\trinamic\M569.cpp:53:22: warning: unused variable 'index' [-Wunused-variable]
   53 |     constexpr int8_t index = -1;
      |                      ^~~~~
Compiling .pio\build\STM32F103RET6_creality\src\src\gcode\geometry\G92.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\gcode\geometry\M206_M428.cpp.o
Marlin\src\gcode\feature\trinamic\M906.cpp: In static member function 'static void GcodeSuite::M906()':
Marlin\src\gcode\feature\trinamic\M906.cpp:60:22: warning: unused variable 'index' [-Wunused-variable]
   60 |     constexpr int8_t index = -1;
      |                      ^~~~~
Compiling .pio\build\STM32F103RET6_creality\src\src\gcode\host\M110.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\gcode\host\M113.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\gcode\host\M114.cpp.o
Marlin\src\gcode\feature\trinamic\M919.cpp: In static member function 'static void GcodeSuite::M919()':
Marlin\src\gcode\feature\trinamic\M919.cpp:97:22: warning: unused variable 'index' [-Wunused-variable]
   97 |     constexpr int8_t index = -1;
      |                      ^~~~~

I tried this HAS_TRINAMIC_CONFIG                    = TMCStepper@0.7.1 in ini/features.ini but don't solve this issue.

ellensp, 2022-03-13:
warnings...  unused variables, safe to ignore them

stikzoo, 2022-03-13:
ok, all was good for compiling.
So when I load my compiled firmware I have a black screen and nothing else.
I did this mod: 
#define E0_DRIVER_TYPE TMC2208_STANDALONE
//#define E0_DRIVER_TYPE TMC2208
And I did not black screen issue.
I saw this issue #22905 but i didn't find any thing.

ellensp, 2022-03-13:
because no one has solved this yet, other than to build with _maple environment  or use different pins IO pins.

You could try this test though, as I have some ideas

You added this to pins_CREALITY_v4.h

```
#if HAS_TMC_UART
  #define E0_SERIAL_TX_PIN                  PA3
  #define E0_SERIAL_RX_PIN                  PA3
  #define TMC_BAUD_RATE                    19200
#endif  
```
As a test try this, see it it makes any difference
```
#if HAS_TMC_UART
  #define E0_SERIAL_TX_PIN                  3
  #define E0_SERIAL_RX_PIN                  3
  #define TMC_BAUD_RATE                    19200
#endif  
```

If nothing changes try editing ini/stm32f1.ini
In block [STM32F103Rx_creality] is
`                              -DSS_TIMER=4 -DTIMER_SERVO=TIM5
`
try

`                              -DSS_TIMER=4 -DTIMER_SERVO=TIM6
`

stikzoo, 2022-03-13:
With STM32RET6_creality_maple I have issue:

- Marlin\src\HAL\STM32F1\onboard_sd.cpp:42:25: error: 'ONBOARD_SD_CS_PIN' was not declared in this scope

With a different pin (PA13) it is working.
I just need to change my soldering pin.
Thank you a lot for your time.

ellensp, 2022-04-23:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22905

closed 2022-04-23 by ellensp