MarlinFirmware/Marlin #23896: [FR] add display sleep

luxflow, 2022-03-13:
### Describe the feature you want

As I know there is "lcd backlight timeout" feature https://github.com/MarlinFirmware/Marlin/pull/23768
And there is "touch screen backlight timeout" feature https://github.com/MarlinFirmware/Marlin/pull/22617

Lcd backlight timeout feature only works when LCD_BACKLIGHT_PIN is defined
I think marlin can leverage u8g sleep feature to implement backlight feature without using additional PIN 
I wrote some POC code for this and tested on ssd1309 lcd (unfortunately I'm not expert in embedded programming nor marlin. sorry in advance)
https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...luxflow:lcd_sleep?expand=1
But problem is that not all devices support u8g sleep feature and there is no function in u8g to check sleep feature in advance (maybe this problem can be solved `#define use_lcd_sleep 1` macro in conditionals_lcd.h)

As I mentioned earlier, there are two backlight timeout for lcd and touch screen
it would be nice to unifying these timeout feature into one universal feature such as `#define DISPLAY_TIMEOUT 30` if possible

I need display sleep feature to save my oled from burn-in

Let me know if I misunderstand something
Thanks for reading my issue

borland1, 2022-03-13:
I would like to see this basic display feature implemented.

Are you assigning a dummy backlight pin in order to enable sleep_on/sleep_off in your pull request code? 

I believe SSD1306 ( also 128x64 OLED) uses same command set as SSD1309.

thisiskeithb, 2022-03-13:
I can see the obvious benefit for OLED screens, but as far as I'm aware, most of the traditional RepRapDiscount full graphic smart controllers available have power wired directly the LCD, so there's no way to turn it off without modding or a new design.

luxflow, 2022-03-14:
> I would like to see this basic display feature implemented.
> 
> Are you assigning a dummy backlight pin in order to enable sleep_on/sleep_off in your pull request code?
> 
> I believe SSD1306 ( also 128x64 OLED) uses same command set as SSD1309.

I forgot to mention how my PR is implemented,
if `LCD_BACKLIGHT_PIN` is defined, use that PIN (current bugfix-2.0.x behavior)
if not defined, use u8g sleep, if device support sleep feature, then it works

To check your device support sleep, check below page, see `U8G_DEV_MSG_SLEEP_ON` is used
https://github.com/MarlinFirmware/U8glib-HAL/tree/master/src/clib 
ssd1306 is supported

I think universal display sleep can be implement one of the following ways 
- (if sleep is supported) u8g sleep
- (if brightness or contrast is supported) brightness or contrast control
- (if pin is defined) additional pin control

borland1, 2022-03-14:
Thanks for the reply, but I still don't understand how you have modifed your the `Configuration.h` or `Configuration_adv.h` file to enable the sleep feature on your SSD1309 display.  

Your above referenced `PR` does not appear to propose modifying the config files, so I assume you are enabling the `TOUCH_IDLE_SLEEP` ( screensaver ) feature with this line:

`#define TOUCH_IDLE_SLEEP 300`

Was expecting to see somthing like this....

```
// OLED Screensaver for displays supporting U8g graphics SLEEP function
#define SCREENSAVER 300
```

then where appropriate, in marlinui.cpp or elsewhere,.. 
```
#if ENABLED( SCREENSAVER ) 
    #define LCD_BACKLIGHT_TIMEOUT
#endif
```

luxflow, 2022-03-15:
I used `#define LCD_BACKLIGHT_TIMEOUT 30` to timeout my screen
I'm not sure what screen type you use, if you're touch screeen it will not works since `LCD_BACKLIGHT_TIMEOUT` only applied in non-touch screen. As I said first in issue, since marlin has 2 implementation for touch, non-touch backlight timeout, have to modify both. My PR dosen't have sleep implementation for TOUCH screen currently..

borland1, 2022-03-15:
I was thinking about testing your PR files code on my printer, which has an OLED display, but since you did not upload your bugfix-2.0.x `Configuration.h` file or since the PR does not modify the `Configuration.h` file, I couldn't figure out how you have the sleep feature enabled.

My custom printer is using a 2.4" OLED ( SSD1309 ), but have Marlin configured as MKS_12864OLED_SSD1306, and menu navigation is with a rotary encoder.  No touch screen.

luxflow, 2022-03-15:
> I was thinking about testing your PR files code on my printer, which has an OLED display, but since you did not upload your bugfix-2.0.x `Configuration.h` and `Configuration_adv.h` files or since the PR does not modify the `Configuration.h` file, I couldn't figure out how you have the sleep feature enabled.
> 
> My custom printer is using a 2.4" OLED ( SSD1309 ), but have Marlin configured as MKS_12864OLED_SSD1306, and menu navigation is with a rotary encoder. No touch screen.

just add `#define LCD_BACKLIGHT_TIMEOUT 10` in `Configuration_adv.h` that's what I changed in my config

borland1, 2022-03-16:
I was able to successfully test your PR code on my printer with OLED display using PlatformIO.  Display sleeps after the set timeout period.  Rotation or a button press of the rotary encoder knob wakes the display from sleep.  Same behavior while printing.

If you don't mind waiting a few days, I can provide code suggestions for implementation with OLED displays to your PR, and also for `Configuration.h`.

borland1, 2022-03-20:
Submitted PR #23927 for review.

thisiskeithb, 2022-03-21:
> Submitted PR https://github.com/MarlinFirmware/Marlin/pull/23927 for review

Closing since there’s now a PR.

closed 2022-03-21 by thisiskeithb