MarlinFirmware/Marlin #23900: Bed leveling - Error: Probing failed (SLOW Probe fail! - No trigger)

yellobello, 2022-03-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi Internet!

I had to change my Bear-PrusaMK3s´s mainboard and decided to go with a "BTT002"-32bit drop-in replacement board for the original Prusa Einsy Rambo board. I prefer Marlin anyway ;-)
Ok, so I took the MK3s- default config from the marlin configs and made it work with the new mainboard. There were some TMC2209-specific entrys which I changed to TMC2130-values like the endstop stall sensitivity, as well as SPI-Endstop settings.

But, my Bed leveling does not work, neither bilinear nor UBL.

I have made a detailled bed level log (M111 S32) and included it. It showas what I have decribed above, so, G28,G29. I also have included my config.h and config_adv.h. 
**The error I keep getting is: Error: Probing failed (more specific: SLOW Probe fail! - No trigger)**

It would be awesome If anybody has an idea whats wrong here.

### Bug Timeline

New bug, but I have seen reports of the same error dissapearing with older marlin versions

### Expected behavior

Normal bed leveling procedure

### Actual behavior

When I do a G28, everything works fine, the head moves to the left and touches the frame, the bed moves to the back and touches the frame, then the head moves to the center of the bed and lowers until the LED on my Pinda probe goes off, so clearly everything related to G28 seems to work.
When doing a G29, however, the bed and head heat up (I configured it that way), the head touches the left frame, then the bed again the back, and then all of a sudden the leveling stops. Both with Bilinear and UBL.

### Steps to Reproduce

G28,G29

### Version of Marlin Firmware

current bugfix and latest release

### Printer model

Bear-Prusa-MK3s

### Electronics

Biq BTT002, TMC2130

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8242869/Configuration.zip)

**Log:**

`Connecting...
Printer is now online.
>>> m111 s32
SENDING:M111 S32
echogrinning smileyEBUG:LEVELING
>>> G28  X0.00 Y-2.20 Z0.20
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X23.00 Y5.00 Z0.30 (Right-Back & Above Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
X current: 500 -> 250
Y current: 600 -> 300
Raise Z (before homing) by 2.00
>>> do_blocking_move_to  X0.00 Y-2.20 Z0.20
>  X0.00 Y-2.20 Z2.00
<<< do_blocking_move_to  X0.00 Y-2.20 Z2.00
>>> homeaxis(X)
Home Fast: -382.50mm
>>> do_homing_move  X0.00 Y-2.20 Z2.00
...(X, -382.50, [50.00])
<<< do_homing_move  X0.00 Y-2.20 Z2.00
>>> set_axis_is_at_home(X)
Axis X home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position= X0.00 Y-2.20 Z2.00 :
<<< set_axis_is_at_home(X)
  current_position= X0.00 Y-2.20 Z2.00 : sync_plan_position
  current_position= X0.00 Y-2.20 Z2.00 : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Home Fast: -324.75mm
>>> do_homing_move  X2.00 Y-2.20 Z2.00
...(Y, -324.75, [50.00])
<<< do_homing_move  X2.00 Y-2.20 Z2.00
>>> set_axis_is_at_home(Y)
Axis Y home_offset = 0.00 position_shift = 0.00
> home_offset[Y] = 0.00
  current_position= X2.00 Y-2.20 Z2.00 :
<<< set_axis_is_at_home(Y)
  current_position= X2.00 Y-2.20 Z2.00 : sync_plan_position
  current_position= X2.00 Y-2.20 Z2.00 : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
>>> home_z_safely  X2.00 Y-0.20 Z2.00
  current_position= X2.00 Y-0.20 Z2.00 : sync_plan_position
  destination= X104.00 Y101.25 Z0.00 : home_z_safely
>>> do_blocking_move_to  X2.00 Y-0.20 Z2.00
>  X104.00 Y101.25 Z2.00
echo:busy: processing
<<< do_blocking_move_to  X104.00 Y101.25 Z2.00
>>> homeaxis(Z)
  current_position= X104.00 Y101.25 Z2.00 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(2.00)
>>> do_blocking_move_to  X104.00 Y101.25 Z2.00
>  X104.00 Y101.25 Z2.00
<<< do_blocking_move_to  X104.00 Y101.25 Z2.00
>>> do_blocking_move_to  X104.00 Y101.25 Z2.00
>  X104.00 Y101.25 Z2.00
<<< do_blocking_move_to  X104.00 Y101.25 Z2.00
Home Fast: -315.00mm
>>> do_homing_move  X104.00 Y101.25 Z2.00
...(Z, -315.00, [5.00])
echo:busy: processing
<<< do_homing_move  X104.00 Y101.25 Z2.00
Move Away: 2.00mm
>>> do_homing_move  X104.00 Y101.25 Z2.00
...(Z, 2.00, 20.00)
<<< do_homing_move  X104.00 Y101.25 Z2.00
Re-bump: -4.00mm
>>> do_homing_move  X104.00 Y101.25 Z2.00
...(Z, -4.00, 10.00)
<<< do_homing_move  X104.00 Y101.25 Z2.00
>>> set_axis_is_at_home(Z)
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> probe.offset.z = 0.30
Axis Z home_offset = 0.00 position_shift = 0.00
> home_offset[Z] = 0.00
  current_position= X104.00 Y101.25 Z-0.10 :
<<< set_axis_is_at_home(Z)
  current_position= X104.00 Y101.25 Z-0.10 : sync_plan_position
  current_position= X104.00 Y101.25 Z-0.10 : > AFTER set_axis_is_at_home
  current_position= X104.00 Y101.25 Z-0.10 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X104.00 Y101.25 Z-0.10
>  X104.00 Y101.25 Z-0.10
<<< do_blocking_move_to  X104.00 Y101.25 Z-0.10
<<< homeaxis(Z)
<<< home_z_safely  X104.00 Y101.25 Z1.90
  current_position= X104.00 Y101.25 Z1.90 : sync_plan_position
  current_position= X104.00 Y101.25 Z1.90 : Leveling OFF
  current_position= X104.00 Y101.25 Z0.00 : ...Now ON
  current_position= X104.00 Y101.25 Z0.00 : sync_plan_position
Restore driver current...
<<< G28  X104.00 Y101.25 Z0.00
>>> G29  X104.00 Y101.25 Z0.00
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X23.00 Y5.00 Z0.30 (Right-Back & Above Nozzle)
Auto Bed Leveling: BILINEAR (enabled)
Z Fade: 10.00
ABL Adjustment Z 0.000 ( 0.000)
Preheating hotend (160) and bed (50)
Wait for bed heating...
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
  current_position= X104.00 Y101.25 Z0.00 : Leveling ON
  current_position= X104.00 Y101.25 Z0.00 : ...Now OFF
  current_position= X104.00 Y101.25 Z0.00 : sync_plan_position
  current_position= X104.00 Y101.25 Z0.00 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(2.00)
>>> do_blocking_move_to  X104.00 Y101.25 Z0.00
>  X104.00 Y101.25 Z2.00
<<< do_blocking_move_to  X104.00 Y101.25 Z0.00
>>> do_blocking_move_to  X104.00 Y101.25 Z0.00
>  X104.00 Y101.25 Z0.00
<<< do_blocking_move_to  X104.00 Y101.25 Z0.00
>>> Probe::probe_at_point  X104.00 Y101.25 Z0.00
...(23.00, 10.00, raise, 0, probe_relative)
  current_position= X104.00 Y101.25 Z0.00 :
>>> do_blocking_move_to  X104.00 Y101.25 Z0.00
>  X0.00 Y5.00 Z0.00
<<< do_blocking_move_to  X0.00 Y5.00 Z0.00
  current_position= X0.00 Y5.00 Z0.00 : Probe::set_deployed
deploy: 1
>>> Probe::run_z_probe  X0.00 Y5.00 Z0.00
>>> Probe::probe_down_to_z  X0.00 Y5.00 Z0.00
>>> do_blocking_move_to  X0.00 Y5.00 Z0.00
>  X0.00 Y5.00 Z-2.30
<<< do_blocking_move_to  X0.00 Y5.00 Z0.00
  current_position= X0.00 Y5.00 Z0.00 : sync_plan_position
<<< Probe::probe_down_to_z  X0.00 Y5.00 Z0.00
SLOW Probe fail! - No trigger.
<<< Probe::run_z_probe  X0.00 Y5.00 Z0.00
  current_position= X0.00 Y5.00 Z0.00 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X0.00 Y5.00 Z0.00
>  X0.00 Y5.00 Z0.00
<<< do_blocking_move_to  X0.00 Y5.00 Z0.00
Errortongue sticking out smileyrobing Failed
[ERROR] Errortongue sticking out smileyrobing Failed

<<< Probe::probe_at_point  X0.00 Y5.00 Z0.00
  current_position= X0.00 Y5.00 Z0.00 : Leveling OFF
  current_position= X0.00 Y5.00 Z0.00 : ...Now ON
  current_position= X0.00 Y5.00 Z0.00 : sync_plan_position
  current_position= X0.00 Y5.00 Z0.00 : Probe::set_deployed
deploy: 0
  current_position= X0.00 Y5.00 Z0.00 : > probing complete
  current_position= X0.00 Y5.00 Z0.00 : sync_plan_position
<<< G29  X0.00 Y5.00 Z0.00`

yellobello, 2022-03-14:
Oh, I forgot:

Some mention that **Z_PROBE_LOW_POINT ** could be involved, so I decreased it from -2 to -5, still the same problem

thisiskeithb, 2022-03-14:
Did you cut/remove the diag pins on Z & E drivers?

yellobello, 2022-03-14:
I just removed all DIAG0,1 pins on all my TMC drivers, but the problem still persists.
At least, homing still works, presumably because of #define SPI_ENDSTOPS.

Yeah, it still does the same thing.

thisiskeithb, 2022-03-14:
> I just removed all DIAG0,1 pins on all my TMC drivers

Why? I only said Z & E.

> homing still works, presumably because of #define SPI_ENDSTOPS.

This is a configuration problem. Disable `SPI_ENDSTOPS` and solder in diag pins for X & Y only.

Please [consult the motherboard’s documentation](https://github.com/bigtreetech/BTT002-V1.0) and use the resources linked in my last comment for support.

yellobello, 2022-03-15:
> Why? I only said Z & E.

My explorative nature led me to experiment a little. I find the idea of using SPI for Endstop related data cool. And I didn´t think it would interfere with my current problem.

> Disable `SPI_ENDSTOPS` and solder in diag pins for X & Y only.

This time I did as you said, but my original problem is still there.
When executing a G29, the print head goes to 0,0,0(or there about) and as soon as it touches the frame, it stops and probing is not executed. Even the light on the pinda probe stays on, so no probing even at 0,0,0 it just stops there.

There must be a reason for this behaviour. What event would lead to `SLOW Probe fail! - No trigger.? `

yellobello, 2022-03-15:
I will be trying some other support places and leave you guys coding ;-)