MarlinFirmware/Marlin #23901: [FR] Support SPINDLE_LASER_FREQUENCY on more platforms

KillerBug, 2022-03-14:
### Is your feature request related to a problem? Please describe.

The default frequency of 2500 is not good for CO2 lasers and there is no way to change it on modern boards.  For example, it does nothing on an Octopus 1.1

### Are you looking for hardware support?

Octopus 1.1

### Describe the feature you want

I would like to be able to set SPINDLE_LASER_FREQUENCY to 20,000 when using an Octopus 1.1

descipher, 2022-03-14:
#define MARLIN_DEV_MODE in configuration.h and it will work. 
FYI 20K is too high for almost every PSU out there.

KillerBug, 2022-03-14:
20000hz is pretty common with CO2 lasers; it's the default on Ruida boards.

I just tested with dev mode turned on, I do get a change in tone when I change the frequency...but the buzzing remains even at 30khz, which shouldn't be.  Wish I had a scope to see if it's actually putting out 30khz.

descipher, 2022-03-15:
What pins are you using?

KillerBug, 2022-03-15:
PD13 for my enable pin and PE11 for my PWM pin.  I feel like at this point it may be more trouble shooting than feature request...been working on it 12 hours now and getting dizzy.  If the functionality exists and is just hidden in dev mode then I guess we are good.  Any idea why it's in dev mode BTW?

closed 2022-03-15 by KillerBug