MarlinFirmware/Marlin #23902: [BUG] Error: Build environment 'STM32F103RET6_creality' is incompatible with BOARD_CREALITY_V427

bharb1, 2022-03-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/8249515/Marlin-bugfix-2.0.x.zip)

I am having issues building my firmware for my new creality V427 in my ender 3. I have been following this [video](https://www.youtube.com/watch?v=neS7lB7fCww&t=891s ) verbatim, including the steps later for configuring the BL touch in the build. I am running into an error when building the firmware which is as follows: 

> Executing task: C:\Users\bharb\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RET6_creality (platform: ststm32; board: genericSTM32F103RE; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Error: Build environment 'STM32F103RET6_creality' is incompatible with BOARD_CREALITY_V427. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple
======================================================== [FAILED] Took 0.92 seconds ========================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:00.917
================================================== 1 failed, 0 succeeded in 00:00:00.917 ================================================== 
The terminal process "C:\Users\bharb\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

This error does not really make sense to me as it seems that this env seems to be the only valid one for this mainboard however the error directly contradicts this.  I have tried some of the suggested envs in the error to see if they would work and I have managed to build with some of them however the printer seems to ignore the firmware when I try to run it. I have been looking at this for a few days now without any progress so any suggestions are appreciated. 

### Expected behavior

Successful build of firmware using env STM32F103RET6_creality

### Actual behavior

Error stating that this env is incompatible with mainboard

### Steps to Reproduce

1. Open Marlin in open studio
2. Open Marlin file in open studio
3. Open config files in open studio
4. Attempt to build firmware 

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Ender 3

### Electronics

BL Touch, V427

### Add-ons

Microswiss Hot end

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

thisiskeithb, 2022-03-15:
```log
Error: Build environment 'STM32F103RET6_creality' is incompatible with BOARD_CREALITY_V427. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple
```

As the error message states, you’ve selected the wrong environment. Use one of those listed in the error message.

closed 2022-03-15 by thisiskeithb

bharb1, 2022-03-15:
> ```
> Error: Build environment 'STM32F103RET6_creality' is incompatible with BOARD_CREALITY_V427. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, [...]

I attempted to do so as stated in my initial message however the printer does not read this firmware as the correct environment (as far as I have gathered) is the one causing the error. Is there something I am missing? The video tutorial used the same env successfully.

ellensp, 2022-03-15:
STM32F103RET6_creality was renamed STM32F103RE_creality
This error is there specifically to tell you STM32F103RET6_creality is no longer a valid option

bharb1, 2022-03-15:
> STM32F103RET6_creality was renamed STM32F103RE_creality This error is there specifically to tell you STM32F103RET6_creality is no longer a valid option

I did try this particular environment however the my printer seems to not recognize this on boot. It just runs the base firmware without updating from the SD card. Perhaps this requires a new post as it is an new issue but may I be missing something in the process of updating the firmware? I figured it was as simple as putting the BIN file on the SD card and starting the printer.

ellensp, 2022-03-15:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

bharb1, 2022-03-15:
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

Noted, I will do so. Sorry, I am new to this community and I thought this was the right place to start.

allanonmage, 2022-05-22:
PlatformIO can't find the problematic part of the code.  If it's a simple text variable change, you should have used an alias.  This is totally a PEBKAC error on behalf of whomever made the decision, not a user support issue.