MarlinFirmware/Marlin #23904: [BUG] 2.0.x nightly build compile error -- manual_feedrate_mm_s

planecrazy29, 2022-03-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Compiling fails with this error:

Compiling .pio\build\BIGTREE_SKR_2_F429\src\src\libs\crc16.cpp.o
Marlin\src\lcd\marlinui.cpp: In static member function 'static void ManualMove::task()':
Marlin\src\lcd\marlinui.cpp:778:60: error: 'manual_feedrate_mm_s' was not declared in this scope; did you mean 'feedrate_mm_s'?
  778 |         const feedRate_t fr_mm_s = (axis < LOGICAL_AXES) ? manual_feedrate_mm_s[axis] : XY_PROBE_FEEDRATE_MM_S;
      |                                                            ^~~~~~~~~~~~~~~~~~~~
      |                                                            feedrate_mm_s

Main branch build with the same configs compiles with no error.

### Bug Timeline

New

### Expected behavior

I expected it to compile

### Actual behavior

It failed to compile

### Steps to Reproduce

Compile with 2.0.x nightly build

### Version of Marlin Firmware

2.0.9.x

### Printer model

Sunlu S8

### Electronics

BTT SKR2 F429

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[CONFIGS.zip](https://github.com/MarlinFirmware/Marlin/files/8255152/CONFIGS.zip)

thisiskeithb, 2022-03-15:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23656

thisiskeithb, 2022-03-15:
Configs must match the version of Marlin you’re compiling for.

closed 2022-03-15 by thisiskeithb