MarlinFirmware/Marlin #23906: Problems with file "pins_BTT_SKR_V2_0_common.h" for EXP1 and EXP2 (Marlin issue)

hcet14, 2022-03-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have issued this bug yesterday here: https://github.com/bigtreetech/SKR-2/issues/131

I'm trying to hook up a "A10S A13 Display" to SKR 2, EXP1.
[image]
[image]
No success so far. Trying to investigate the problem leaded me to the differences of the "pins_BTT_SKR_V2_0_common.h" file.

I think the pins should correlate to the schematics. I checked the schematics https://github.com/bigtreetech/SKR-2/blob/master/Hardware/BIGTREETECH%20SKR%202-SCH.pdf with my multimeter, schematics are fine regarding EXP1 and EXP2.

https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src/pins/stm32f4/pins_BTT_SKR_V2_0_common.h
[image]
#define EXP1_03_PIN                         PE13
#define EXP1_04_PIN                         PE12
#define EXP1_05_PIN                         PE11
#define EXP1_06_PIN                         PE10
#define EXP1_07_PIN                         PE9
#define EXP1_08_PIN                         PB1
#define EXP1_09_PIN                         PB0
#define EXP1_10_PIN                         PC5

#define EXP2_03_PIN                         -1
#define EXP2_04_PIN                         PC4
#define EXP2_05_PIN                         PA7
#define EXP2_06_PIN                         PB2
#define EXP2_07_PIN                         PA4
#define EXP2_08_PIN                         PE7
#define EXP2_09_PIN                         PA5
#define EXP2_10_PIN                         PA6

https://github.com/bigtreetech/SKR-2/blob/master/Firmware/Marlin-bugfix-2.0.9.2.x/Marlin/src/pins/stm32f4/pins_BTT_SKR_V2_0_common.h
[image]
#define EXP1_03_PIN                         PE13
#define EXP1_04_PIN                         PE12
#define EXP1_05_PIN                         PE11
#define EXP1_06_PIN                         PE10
#define EXP1_07_PIN                         PE9
#define EXP1_08_PIN                         PB1
#define EXP1_09_PIN                         PB0
#define EXP1_10_PIN                         PC5

#define EXP2_03_PIN                         -1
#define EXP2_04_PIN                         PC4
#define EXP2_05_PIN                         PA7
#define EXP2_06_PIN                         PB2
#define EXP2_07_PIN                         PA4
#define EXP2_08_PIN                         PE7
#define EXP2_09_PIN                         PA5
#define EXP2_10_PIN                         PA6

hcet14
[image]
#define EXP1_01_PIN                         PC5
#define EXP1_02_PIN                         PB0
#define EXP1_03_PIN                         PB1
#define EXP1_04_PIN                         PE9
#define EXP1_05_PIN                         PE10
#define EXP1_06_PIN                         PE11
#define EXP1_07_PIN                         PE12
#define EXP1_08_PIN                         PE13

#define EXP2_01_PIN                         PA6
#define EXP2_02_PIN                         PA5
#define EXP2_03_PIN                         PE7
#define EXP2_04_PIN                         PA4
#define EXP2_05_PIN                         PB2
#define EXP2_06_PIN                         PA7
#define EXP2_07_PIN                         PC4
#define EXP2_08_PIN                         -1

I'm pretty sure, mine is the right one. 

Any help is appreciated! 

### Bug Timeline

New 2022-03-14

### Expected behavior

I'm a bloody beginner with marlin. I think this is a bug. I hope there will be a feedback from someone more experienced than me.

### Actual behavior

No feedback yet.

### Steps to Reproduce

It is not reproducible.

### Version of Marlin Firmware

Marlin 2.0.9.3

### Printer model

CTC A13 with a lot of new parts

### Electronics

BTT SKR 2 with STM32F429

### Add-ons

original "A10S A13 Display" based on ST7920 connected by SPI

### Additional information & file uploads

This is my "pins_BTT_SKR_V2_0_common.h"
[pins_BTT_SKR_V2_0_common.zip](https://github.com/MarlinFirmware/Marlin/files/8256291/pins_BTT_SKR_V2_0_common.zip)
I made a special cable to connect the 10-pin-heathers on "SKR 2" and "A10S A13 Display".
Here is the pinning:
[pinning_display_skr2_marlin_issue.ods](https://github.com/MarlinFirmware/Marlin/files/8256380/pinning_display_skr2_marlin_issue.ods)
Backlight works.
The display works, when connected to my old mainboard (Anet 3D) with a 1:1 cable.

thisiskeithb, 2022-03-15:
You simply need to rotate cables/EXP housings for some LCDs since pins can be 180 out depending on the OEM of the LCD.

closed 2022-03-15 by thisiskeithb

hcet14, 2022-04-16:
I got it [https://github.com/bigtreetech/SKR-2/issues/131#issuecomment-1100735095](url). Sorry for wasting your time.

Kind regards, hcet14