MarlinFirmware/Marlin #23909: [BUG] Compiling error (reproducable and avoidable)

TeslorKaalut, 2022-03-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After a lot of useless tries to compile the actual release, also the bugfix version i finally found out what causes the autocompiler in platform.io to not compile correctly. No matter if i use actual release, or bugfix version of 2.0.x, its the same result every time:

```
Processing mega2560 (board: megaatmega2560; platform: atmelavr@~3.4; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal_AIP31068> 1.0.4
|   |-- <SoftSPIB> 1.1.2
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <SoftSPIB> 1.1.2
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.5.1
|   |-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fast_pwm.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fastio.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\HAL_MinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\mega2560\src\src\MarlinCore.cpp.o
Compiling .pio\build\mega2560\src\src\core\serial.cpp.o
Compiling .pio\build\mega2560\src\src\core\utility.cpp.o
Compiling .pio\build\mega2560\src\src\feature\ammeter.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M220.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M221.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M301.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M302.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M309.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M92.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M108_M112_M410.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M111.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M120_M121.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M17_M18_M84.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M211.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M400.cpp.o
[... 232 lines cut ...]
Geeetech i3ProB

### Electronics

Stock

### Add-ons

N/A

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

Other (explain below)

### Additional information & file uploads

for normal single gcode commands i use pronterface and easyprint3d, i print via sd card and the firmware was uploaded with platform.io, after compiling

ellensp, 2022-03-17:
Never wind worked out what your talking about, even without you mentioning key things like your lcd type

this error occurs with YHCB2004

ellensp, 2022-03-17:
According to the example configuration Geeetech i3ProB has a REPRAP_DISCOUNT_SMART_CONTROLLER on a BOARD_GT2560_REV_A_PLUS, not a YHCB2004 

https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.0.x/config/examples/Geeetech/Prusa%20i3%20Pro%20B/noprobe/Configuration.h

ellensp, 2022-03-17:
The YHCB2004 has 1 20 pin conenctor, the BOARD_GT2560_REV_A_PLUS has 2 * 10 pin connectors

Are you shore you have the correct hardware? set?

ellensp, 2022-03-17:
Please attache pictures of motherboard and the lcd display (front and back)

TeslorKaalut, 2022-03-17:
> Never wind worked out what your talking about, even without you mentioning key things like your lcd type
> 
> this error occurs with YHCB2004 on motherboard that are not setup for this LCD (ie all motherboard other than >BOARD_GT2560_V3)

Well be asured that i know just enough to create real desasters and thanks for that kind reply.... pretty shure this will motivate a lot of people to share there knowing here....

i have to admit that i was a bit divertet and the lines here confused me:

```
//
// GT2560 (YHCB2004) LCD Display 
// 
// Requires Testato, Koepel softwarewire library and 
// Andriy Golovnya's LiquidCrystal_AIP31068 library.
// 
// #define YHCB2004
```
i saw the boardname (without any revision detail and the 2004 was there, so i didnt had any idea to think twice about. about the rest you were right, its the REPRAP_DISCOUNT_SMART_CONTROLLER but with a green pcb 

here are the requestet photos, i hope they help you (i didnt ask a thing, just wanted to help, so maybe ignore me would have been a bit more polite?) when im home from work ill try with the appropriate definition HAND :) 

[image]
[image]

ellensp, 2022-03-17:
That doesn't look like a  YHCB2004 to me. Disable  YHCB2004 and enable REPRAP_DISCOUNT_SMART_CONTROLLER

TeslorKaalut, 2022-03-17:
> That doesn't look like a YHCB2004 to me. Disable YHCB2004 and enable REPRAP_DISCOUNT_SMART_CONTROLLER

Nice idea.... thank you that you give me the tipp to deaktivate the wrong definition (what i did as written in the first post
and even more genius is your tipp to aktivate the definition that i described in my first answer to your (what four? five?) one scentence replies because i didnt write just one line above your post that ill do that when coming home....

sorry, but my post might be useless to you, ok.... but as i didnt ask anything i see no reason that your tone is inpolite and in addition to that you obviusly didnt read anything i wrote except the title and the code part and looked at the photos.... kinda weird way to treat some (yes again, maybe useless) try to help others, really

closed 2022-03-17 by ellensp