MarlinFirmware/Marlin #23910: [BUG] Error:Field planner_leveling_active mismatch

mnul, 2022-03-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Flashing the latest Bugfix 2.0 is giving me an EEPROM error. 
Reset (`M502`) works, but saving to EEPROM (`M500`) gives me an `EEPROM Datasizve error`.

Turning on `#define DEBUG_EEPROM_READWRITE` prints the error as `Error:Field planner_leveling_active mismatch`.

### Steps to Reproduce

1. Build Marlin bugfix 2.0
2. Flash Marlin bugfix 2.0
3. M502
4. M500
5. EEPROM error

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Mar 17 2022 10:20:25)

### Printer model

Creality Ender 5

### Electronics

SKR V1.3, TMC2209, BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8283565/Configuration.zip)

ellensp, 2022-03-17:
Issue is with X_AXIS_TWIST_COMPENSATION eeprom settings, creating a pr

thisiskeithb, 2022-03-17:
See https://github.com/MarlinFirmware/Marlin/issues/23744

closed 2022-03-17 by thisiskeithb