MarlinFirmware/Marlin #23912: [BUG] CoreXY only one stepper moving

niklasber, 2022-03-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I updated from version 2.0.7.2 to the newest 2.0.9.3 and without changing anything in the code, one stepper motor of my CoreXY configuration doesn't work anymore. It just seems to not getting any signal to move. By going back to the older firmware this issue is fixed again.
Do I need to change anything in the code in the new version? 

### Version of Marlin Firmware

2.0.9.3

thisiskeithb, 2022-03-17:
There was an issue opened where coreXY became coreYX, but it sounds like things have gotten worse: https://github.com/MarlinFirmware/Marlin/issues/23747

ellensp, 2022-03-18:
What motherboard?

niklasber, 2022-03-18:
Ramps 1.4

ellensp, 2022-03-18:
Please attach your configuration files.

niklasber, 2022-03-18:
[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/8302530/Archiv.zip)

ahmetcemturan, 2022-03-18:
You have CoreYX instead of CoreXY.   maybe thats it?

//#define COREXY
//#define COREXZ
//#define COREYZ
#define COREYX
//#define COREZX
//#define COREZY

niklasber, 2022-03-18:
No that’s right because of my stepper layout. Wouldn‘t explain that Motor isn‘t moving neither.

ellensp, 2022-03-19:
I just set up a ramps. with the same display and a4988 stepper drivers. but on 12v 
Using your configuration files (update the compile on bugfix)
This works as expected
In absolute mode
I tested with G1 X200, both X and Y stepper driver move as expected
I then tested with G1 X0  and both X and Y steper drivers move as expected in the opposite direction
In relative mode
G1 X50
G1 X-50
All works as expected

ellensp, 2022-03-19:
Best guess is your ramps is being used in a non standard way. Where the stepper drivers have been swapped around.
Verify which stepper motor wires goes to which stepper driver on the ramps.

niklasber, 2022-03-19:
Hmm okay, thank you. I noticed that there are now 2 stepper drivers per axis (x and x2 für example) how do I handle that? The ramps has just one connection per axis right?
One of the motors is connected to the ramps on the X driver and the other with the Y. 
With the older Marlin Version it works fine

ellensp, 2022-03-19:
My other guess is you have config files in wrong place and what you attached  (which works fine) is not the config files being used to compile and upload Marlin 

Try a simple change #define CUSTOM_MACHINE_NAME "testing"  
save, compile and upload 

It should now say "testing  bereit" on the main status screen bottom line

niklasber, 2022-03-19:
Thank you, I will try this but I doubt that’s gonna help because the stepper moves when I try my old config, so it is definitely using this one.

thisiskeithb, 2022-03-19:
Since your configs had to be ported to work correctly, make sure you are using the latest [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.0.x.zip)

ellensp, 2022-03-19:
it only needed the standard `#if ANY(HAS_LCD_MENU, EXTENSIBLE_UI, HAS_DWIN_E3V2)
` to `#if HAS_MANUAL_MOVE_MENU` line change to compile.

niklasber, 2022-03-31:
I tried it with a lot more Marlin versions now and every single version from 2.0.8 up to the newest only move one of the stepper motors. Once back to 2.0.7.2 everything works fine without changing any software or hardware. Someone knows if there was a big change since 2.0.8 with the stepper or the core configuration?

ellensp, 2022-04-01:
I have just done some corexy testing on a old ramps and two a4988 stepper driver and stepper motors on X and Y
Tests where done in absolute more (ie power on state) 

```
    Gcode Command  | G1 X50                 | G1 X0                           | G1 Y50                     | G1 Y0                      |
| version | mode   | X(a)      | Y(b)       | X(a)           | Y(b)           | X(a)      | Y(b)           | X(a)           | Y(b)      |
| bugfix  | corexy | clockwise | clockwise  | anti clockwise | anti clockwise | clockwise | anti clockwise | anti clockwise | clockwise |
| 2.0.9.3 | corexy | clockwise | clockwise  | anti clockwise | anti clockwise | clockwise | anti clockwise | anti clockwise | clockwise |
| 2.0.8.2 | corexy | clockwise | clockwise  | anti clockwise | anti clockwise | clockwise | anti clockwise | anti clockwise | clockwise |
| 2.0.7.2 | corexy | clockwise | clockwise  | anti clockwise | anti clockwise | clockwise | anti clockwise | anti clockwise | clockwise |
| 2.0.6.1 | corexy | clockwise | clockwise  | anti clockwise | anti clockwise | clockwise | anti clockwise | anti clockwise | clockwise |
| 2.0.5.4 | corexy | clockwise | clockwise  | anti clockwise | anti clockwise | clockwise | anti clockwise | anti clockwise | clockwise |
| 2.0.4.4 | corexy | clockwise | clockwise  | anti clockwise | anti clockwise | clockwise | anti clockwise | anti clockwise | clockwise |
| 2.0.3.0 | corexy | clockwise | clockwise  | anti clockwise | anti clockwise | clockwise | anti clockwise | anti clockwise | clockwise |
| 2.0.0.0 | corexy | clockwise | clockwise  | anti clockwise | anti clockwise | clockwise | anti clockwise | anti clockwise | clockwise |
| 1.1.9   | corexy | clockwise | clockwise  | anti clockwise | anti clockwise | clockwise | anti clockwise | anti clockwise | clockwise |

```

No issues where found

ellensp, 2022-04-01:
Try swapping X and Y stepper drivers, does fault stay or move axis?
Also try another mega2560

Your provided configs still work fine and non one else has this issue.

niklasber, 2022-04-01:
So I swapped the cables on the ramps and it is still the same motor moving. Before it was connected to y now to x. But I cannot imagine the motor or stepper driver to be broken since the function with Marlin older than 2.0.8…

ellensp, 2022-04-01:
"So I swapped the cables on the ramps " when I asked you to swap stepper drivers not motors...

niklasber, 2022-04-01:
Sorry.. I changed stepper drivers now and the other motor starts rotating, so it changes with the driver. the other driver must be holding the whole time because to doesn't turn one of the motors no matter if x or y. And when I unplug the driver the motor doesn't get hold anymore and turn from the belt movement of the other motor.

ellensp, 2022-04-01:
so you have a dying stepper driver.  What ever changed happened in firmware now triggers this hardware  issue.

closed 2022-04-01 by ellensp