MarlinFirmware/Marlin #23917: [BUG] Error when trying to compile for LPC1768 (MOTHERBOARD BOARD_MKS_SGEN_L)

ahmetcemturan, 2022-03-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I get an error when trying to compile.
Error is pasted below in actual behaviour.

I found this:   https://github.com/MarlinFirmware/Marlin/issues/18102
But it is a resolved bug but did not find that addition in the curren HAL  file..

### Bug Timeline

New

### Expected behavior

To compile

### Actual behavior

Gives following Error:

`Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\fast_pwm.cpp.o
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/MarlinSerial.h:25,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:44,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
C:\Users\Ahmet\.platformio\packages\framework-arduino-lpc176x\cores\arduino/WString.h:37: warning: "F" redefined
   37 | #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
      | 
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/../shared/Marduino.h:89,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:39,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/../shared/progmem.h:42: note: this is the location of the previous definition
   42 | #define F(str) (reinterpret_cast<const __FlashStringHelper *>(PSTR(str)))
      | 
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/MarlinSerial.h:25,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:44,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL_MinSerial.cpp:24:
C:\Users\Ahmet\.platformio\packages\framework-arduino-lpc176x\cores\arduino/WString.h:37: warning: "F" redefined
   37 | #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
      | 
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/../shared/Marduino.h:89,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:39,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL_MinSerial.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/../shared/progmem.h:42: note: this is the location of the previous definition
   42 | #define F(str) (reinterpret_cast<const __FlashStringHelper *>(PSTR(str)))
      | 
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/MarlinSerial.h:25,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:44,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
C:\Users\Ahmet\.platformio\packages\framework-arduino-lpc176x\cores\arduino/WString.h:37: warning: "F" redefined
   37 | #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
      | 
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/../shared/Marduino.h:89,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:39,
                 from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
[... 287 lines cut ...]
.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Prusa

### Electronics

MKS SGEN L

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

ellensp, 2022-03-19:
Cannot reproduce with just setting #define MOTHERBOARD BOARD_MKS_SGEN_L

ellensp, 2022-03-19:
your using .platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\9.3.1 
mine is .platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\9.2.1 and does not have this cwchar

ellensp, 2022-03-19:
Updated to
 - framework-arduino-lpc176x 0.2.9 
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)

Still cannot replicate

ellensp, 2022-03-19:
Delete your directory c:\users\ahmet\.platformio and restart vscode
This forces a full reinstall of platfomio and required libraries

ahmetcemturan, 2022-03-19:
Deleting the directory solved it.    Note though that it Vstudio compiled fine for Robin Nano board.  Problem was only for 1768 and 1769 (BTT SKR 1.4 Turbo) boards.

Thank You very much @ellensp

closed 2022-03-19 by ellensp