MarlinFirmware/Marlin #23921: Ender 2 Pro W/CR-Touch

The300lbGorilla, 2022-03-19:
### Is your feature request related to a problem? Please describe.

No

### Are you looking for hardware support?

Ender 2 Pro
Creality Board 4.2.3 With Silent Steppers
Need support for CR-Touch ABL Probe

### Describe the feature you want

I am looking for a firmware version (firmware.bin) that has the ABL Probe to function in a 3x3 or 4x4.
Also, if you happen to know where I can get the source code for the Ender 2 Pro, that would be helpful.

### Additional context

Sorry if I sound a bit green, but I have not done any programming since the early 2013, and that was database programming in VB.
Any help would be greatly welcomed. I am trying to get this printer up and running.

thisiskeithb, 2022-03-19:
You'll need to compile your own firmware as we don't provide firmware binaries here. You can use the existing [Ender-2 Pro config](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-2%20Pro/CrealityV423) as a starting point.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-03-19 by thisiskeithb

The300lbGorilla, 2022-03-19:
YOu actually sent me where I needed to go. Configurations. Thank you