MarlinFirmware/Marlin #23922: [BUG] "PIDTEMPBED" doesn't work with BTT SKR v1.4 Turbo (LPC1769)

MakerMeik, 2022-03-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I enable PIDTEMPBED in the configuration.h by activating "#define PIDTEMPBED", my printer my printer cannot reach the demanded bed temperature. The temperature rises and always stops rising at 10°C below the target temperature.
So if I demand 53°C it stops at 43°C. 65°C ends at 55°C and so on.

After a while, Marlin aborts with an error message saying it was unable to reach the temperature (I forget the exact wording).

If I disable PIDTEMPBED everything works as expected.

### Bug Timeline

I tested this feature a few days ago for the first time

### Expected behavior

I expected PIDTEMPBED to work properly with my printer.

### Actual behavior

The printer always stops when trying to reach the requested bed temperature.

### Steps to Reproduce

1. activate "#define PIDTEMPBED"
2. try to start a print
3. wait until you see, that the bed-temperature is not reached

### Version of Marlin Firmware

2.0.9.3 and BugFix-2.0.x

### Printer model

Bigtreetech SKR 1.4 turbo on an rebuilt Anet A8

### Your Slicer

Cura

### Host Software

OctoPrint

ellensp, 2022-03-20:
and your settings for PID, like ? 
#define BANG_MAX 255     // Limits current to nozzle while in bang-bang mode; 255=full current
#define PID_MAX BANG_MAX // Limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full

ellensp, 2022-03-20:
Did you enable PID_BED_DEBUG? and log the output?

MakerMeik, 2022-03-20:
"#define MAX_BED_POWER 255" was already set.
Now I have also enabled PID_BED_DEBUG. Please find the attached log.

[serial(3).log](https://github.com/MarlinFirmware/Marlin/files/8311087/serial.3.log)

ellensp, 2022-03-20:
Are the bed pid values set the same as in the Config file?

MakerMeik, 2022-03-20:
yes, that's the exact corresponding config files.