MarlinFirmware/Marlin #23926: [BUG] MKS-GEN_L (V 1.0) pins config error for double extruder setup

aegelsky, 2022-03-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

when using in Configuration.h:

#define MOTHERBOARD BOARD_MKS_GEN_L
#define EXTRUDERS 2

it uses pins_MKS_GEN_L.h, which in turn uses pins_RAMPS.h,
where FET_ORDER_EEB is selected (which is right), but pins in RAMPS and MKS GEN L don't match in second extruder:

#elif FET_ORDER_EEB                               // Hotend, Hotend, Bed
  #define HEATER_1_PIN              MOSFET_B_PIN
  #define HEATER_BED_PIN            MOSFET_C_PIN

#elif FET_ORDER_EEB                             // Hotend, Hotend, Bed
    #define FAN_PIN                            4  // IO pin. Buffer needed

(from pins_RAMPS.h)

this leads to:
HEATER_1_PIN              = 9 
FAN_PIN                        = 4

and here is actual MKS GEN L pins map:
https://github.com/makerbase-mks/MKS-GEN_L/blob/master/hardware/MKS%20Gen_L%20V1.0_008/MKS%20Gen_L%20V1.0_008%20PIN.pdf

pin D9  is fan pin and D4 is servo pin, so when enabling second extruder heater, fan turns ON and there is no way to power on fan.
(also I wouldn't power up extruder heater from fan socket, it very possibly might no handle 40W output)

adding this into pins_MKS_GEN_L.h (from line 40) fixes this issue:

// Hotend, Hotend, Bed + Fan on D9
#if FET_ORDER_EEB                             
  #define MOSFET_B_PIN  7
  #define FAN_PIN       9
#endif

just trying to help the community.

### Bug Timeline

old (I guess)

### Expected behavior

adding MKS GEN L motherboard uses valid pins

### Actual behavior

pins for extruder 2 and fan are messed up

### Steps to Reproduce

#define MOTHERBOARD BOARD_MKS_GEN_L
#define EXTRUDERS 2

this leads to:
HEATER_1_PIN              = 9 
FAN_PIN                        = 4

which is wrong

### Version of Marlin Firmware

2.0.x (2.0.9.3)

### Printer model

Prusa i3

### Electronics

MKS GEN L V 1.0

### Add-ons

second extruder

### Additional information & file uploads

adding this into pins_MKS_GEN_L.h (from line 40) fixes this issue:

// Hotend, Hotend, Bed + Fan on D9
#if FET_ORDER_EEB                             
  #define MOSFET_B_PIN  7
  #define FAN_PIN       9
#endif

just trying to help the community.

ellensp, 2022-03-20:
Want to try you hand at a PR?  https://marlinfw.org/docs/development/getting_started_pull_requests.html

aegelsky, 2022-03-21:
> Want to try your hand at a PR? https://marlinfw.org/docs/development/getting_started_pull_requests.html

will try to find time for it in next few days..

aegelsky, 2022-03-29:
> Want to try your hand at a PR? https://marlinfw.org/docs/development/getting_started_pull_requests.html

I haven't used git for quite a long time, but looks like I finally made it here:
https://github.com/MarlinFirmware/Marlin/pull/23965

ellensp, 2022-03-30:
PR looks fine. So closing this issue.