MarlinFirmware/Marlin #23932: Compilation erro with MMU2 enable

kursatu, 2022-03-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am getting compilation with MMU2 enabled.
I've fixed it. It was due to trivial mistake in mmu2.cpp
The fix was.
Commenting out
//inline void ATTN_BUZZ(const bool two=false) { BUZZ(200, 404); if (two) { BUZZ(10, 0); BUZZ(200, 404); } }
and removing the argument in this file 
      ATTN_BUZZ(true); 
      to 
      ATTN_BUZZ();

### Bug Timeline

New bug

### Expected behavior

The compilation would succeed.

### Actual behavior

The compilation would fails.

### Version of Marlin Firmware

bugfix-2.0.x

thisiskeithb, 2022-03-21:
Looks like it was broken in https://github.com/MarlinFirmware/Marlin/commit/89a9c3a391101e4d2d8dbfbf0cdb261ad2d9592b

ellensp, 2022-03-21:
yip, conflicts with #define ATTN_BUZZ() do{ BUZZ(198, 404); BUZZ(4, 0); BUZZ(198, 404); }while(0)  in buzzer.h
Adding 
#undef ATTN_BUZZ
above  
inline void ATTN_BUZZ(const bool two=false) { BUZZ(200, 404); if (two) { BUZZ(10, 0); BUZZ(200, 404); } }
gets around it, but isn't  very nice,,,

ellensp, 2022-03-25:
PR created, closing

closed 2022-03-25 by ellensp