MarlinFirmware/Marlin #23933: [BUG] TMC2660 TMC Connection error

alexxy, 2022-03-21:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Printer uses board Azteeg GT5 with tmc2660
Seems like with recent marlin it doesnt work. Marlin cannot connect to drivers. Also there are strange vsense  value (in config it set to 0.10)

Send: M122
Recv: 		X	Y	Z	E	E1
Recv: Enabled		false	false	false	false	false
Recv: Set current	1200	1200	1400	1200	1200
Recv: RMS current	970	970	1141	970	970
Recv: MAX current	1368	1368	1609	1368	1368
Recv: Run current	16/31	16/31	19/31	16/31	16/31
Recv: Hold current
Recv: CS actual
Recv: PWM scale
Recv: vsense		0=.310	0=.310	0=.310	0=.310	0=.310
Recv: stealthChop
Recv: msteps		16	16	16	16	16
Recv: interp
Recv: tstep
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn
Recv: off time	0	0	0	0	0
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv:  -end		-1	-1	-1	-1	-1
Recv:  -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: uStep count
Recv: DRVSTATUS	X	Y	Z	E	E1
Recv: stst		*	*	*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 		X	0x00:00:00:00	 Bad response!
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 		E1	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... Error: All LOW
Recv: Testing E1 connection... Error: All LOW
Recv: ok

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin 2.0.9.3 (Mar 21 2022 14:30:41) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3ntPro EXTRUDER_COUNT:2 UUID:5777a60b-88ed-4102-babe-970ba40c4704

ellensp, 2022-04-01:
no configs, hasn't tried bugfix.  Impossible to even try and help

alexxy, 2022-04-02:
Bugfix have same issue. 
Configs attached

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8402063/Configuration.zip)

ellensp, 2022-04-02:
Himm,   reading through code and example non marlin configs configs spi I see for example

```
motor_driver_control.alpha.spi_channel       0                # SPI channel 1 is sdcard channel
motor_driver_control.alpha.spi_cs_pin        0.10             # SPI CS pin
#motor_driver_control.alpha.spi_frequency     100000          # SPI frequency

and later I see 
panel.spi_channel                            0                 # set spi channel to use P0_18,P0_15 MOSI,SCLK

```
So we want to use those pins for SPI

The CS pin are correct in the boards pins.h

But the SPI pins default to 

#define BOARD_SPI2_SCK_PIN      P0_07
#define BOARD_SPI2_MISO_PIN     P0_08
#define BOARD_SPI2_MOSI_PIN     P0_09

This is not right for your board  (from what little documentation I can find)

So open up Marlin/src/HAL/LPC1768/include/SPI.h

Find the line `#define BOARD_NR_SPI 2`
and change it to 
`#define BOARD_NR_SPI 1`

now it should use the SPI pins

#define BOARD_SPI1_SCK_PIN      P0_15
#define BOARD_SPI1_MISO_PIN     P0_17
#define BOARD_SPI1_MOSI_PIN     P0_18

As expected by your controller.

alexxy, 2022-04-02:
Hi!

I'll try this monday. Seems like it was changed sometimes ago, since 2.0.4 works with my configs...

alexxy, 2022-04-04:
> 

I tryed this. Still have tmc connection error

```
Driver registers:
    X       0x00:00:00:00    Bad response!
    Y       0x00:00:00:00    Bad response!
    Z       0x00:00:00:00    Bad response!
    E       0x00:00:00:00    Bad response!
    E1      0x00:00:00:00    Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW
Testing E1 connection... Error: All LOW

```

closed 2022-06-14 by github-actions[bot]