MarlinFirmware/Marlin #23934: [BUG] Motion inconsistencies in axis A

TimotF, 2022-03-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

## Notes on hardware
I am using a stepper motor with a gearbox 1:10 (output is 10 times slower than the motor shaft). All hardware is very high quality and cannot be the cause of this issue. 
## Behavior
#### Behavior after configuring this motor on axis Y (same exact motor and endstops configuration but on axis Y)
- It is working just fine. There is no deviation or offset at any time...
#### Behavior after configuring this motor on axis B ( second additional linear axis)
- After homing, a very small offset can appear on this axis, causing dimensional accuracy problems in the resulting object (we need extreme precision). This offset or error only seems to appear after starting a gcode file.
- It is not clear to me if this is a constant offset, or some sort of accumulated error over time.
- The behavior is quite strange: 
  - It is repeatable when sending Gcode **in sequence** -> if I send the exact same Gcode (doesn't matter if gcode is on sd card or send through a USB host), I get the exact same output (resulting object)
  - When we skip part of a Gcode (for ex. to isolate 1 or 2 moves), we don't get the same output as if it was produced in the sequence.
  - When manually setting a position, our measurement tools shows that the position is repeatable, even after moving the axis a few dozen times.
  - We tested extensively our homing sensor, it is accurate with no doubts.
## TL:DR
We observed motion accuracy problems on axis B while on axis Y it is OK. When using axis B, the offset / accumulated error is fairly small, but does matter in our use case.
### Question => Is motion handled any differently between axis Y and axis B?

### Version of Marlin Firmware

latest bugfix (commit 10ce88f810f08675140edb6cf248c87ee549855e)

### Printer model

Custom robot

### Electronics

Using BTT octopus V1.1, TMC2209 drivers

### Additional information & file uploads

The tests have been done on a fork of this repository. You can find [it there, branch RB-Artero](https://github.com/Winnove/Marlin/tree/RB-Artero).

DerAndere1, 2022-03-27:
Just to make sure I understand correctly:
> When manually setting a position, our measurement tools shows that the position is repeatable, even after moving the axis a few dozen times.

Is the position repeatably offset or repeatably correct?

>  This offset or error only seems to appear after starting a gcode file.

1. Would be nice if you could enable `TMC_DEBUG` and `M114_DETAIL`. 
2. How big is the offset the same if you run a gcode file that only contains the following? (if you oberve an offset, please attach the log from the serial output):
```
G28
M112
M114 D
G1 B40
G1 B80
M400
M114 D
```

3. Is the offset the same if you send above commands manually, (after waiting for each move to complete)? What is the output of the final `M114 D`?

> When we skip part of a Gcode (for ex. to isolate 1 or 2 moves), we don't get the same output as if it was produced in the sequence.

If above tests could not reproduce the error, please attach a zip-compressed gcode file (as short as possible) that reproduces the error. Is the position reported by Firmware also wrong?

DerAndere1, 2022-03-27:
Additional Ideas:
- Maybe the observed differences between B and Y axis are due to different settings of  `DEFAULT_AXIS_STEPS_PER_UNIT` and `DEFAULT_MAX_ACCELERATION`. That would indicate an (inevitable?) rounding error is the cause.
- Try experimenting with  `TMC_HOME_PHASE`

DerAndere1, 2022-03-31:
Thanks for the report and sorry for the inconvenience. BACKLASH_COMPENSATION, which probably does not yet work on axes beyond XYZ, could have caused similar behaviour, but it is not enabled in the clients' configs.  So far I could not reproduce with A4988 drivers.

DerAndere1, 2022-04-02:
maybe related to G2 commands (`ARC_SUPPORT` enabled), see https://github.com/MarlinFirmware/Marlin/issues/22461

TimotF, 2022-04-28:
After some time, I finally found the source of this issue. It was actually a hardware fault on the controller board I was using, the BTT Octopus V1.1. For anyone interested, [here is the corresponding issue](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues/114) that allowed me to solve my problem.

Thank you very much for your time @DerAndere1 !

closed 2022-04-28 by TimotF