MarlinFirmware/Marlin #23939: [BUG] Hotend heater in BANG BANG instead of the activated PID

CKoblinger, 2022-03-24:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

My hot ends temp is fluctuating (on 260°C) like +/- 5°C, PID Tuning also seems not to work. Please HELP ME

I've tried out a lot of things during this problem so I can say that marlin is in bang-bang mode and not in the activated PID mode. I've figured out that when I set the current Limit for bang-bang mode to for example "5", as the result my heater is not really heating up like it should (PID_MAX is set to 255).

### Bug Timeline

It started after installing Marlin 2.0.7 (x)

### Expected behavior

I expect that my hotend heats up like normal.

### Actual behavior

The hotend heats up and when it reaches the temp. target it overshoots like 5°C

### Steps to Reproduce

-set a temp value
-watch the temp graph

### Version of Marlin Firmware

2.0.7

### Printer model

ANET E12

### Electronics

SKR 1.4 Turbo

### Add-ons

-

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration + Confi_adv.zip](https://github.com/MarlinFirmware/Marlin/files/8340462/Configuration.%2B.Confi_adv.zip)
[image]

thisiskeithb, 2022-03-24:
> It started after installing Marlin 2.0.7

2.0.7 was released September 2020. We’ve had several releases since with `2.0.9.3` being  the latest version.

> Did you test the latest bugfix-2.0.x code?
> No, but I will test it now!

Please download `bugfix-2.0.x` and test again.

CKoblinger, 2022-03-24:
Hello sorry that I am answering so late but I had to bring the marlin 2.0.7 settings in the new one (2.0.9.3) version. The problem is like before so I don't know what to do jet.

rq3, 2022-03-24:
> Hello sorry that I am answering so late but I had to bring the marlin 2.0.7 settings in the new one (2.0.9.3) version. The problem is like before so I don't know what to do jet.

This looks very much like a PID mis-tune (not a bang-bang problem). Try setting the P value at half of what it is now in configuration.h, and check again. Let us know what happens, and don't leave the printer un-attended while doing this. Also, make very sure that you don't have a loose heater or thermistor in your hot end.

CKoblinger, 2022-03-24:
looks like that was a problem with a pid setting, it is stupid that when I make a pid autotune it doesn't work. Tanks for the help

closed 2022-03-25 by thisiskeithb