MarlinFirmware/Marlin #23941: Zonestar LCD buttons not working[BUG] (Mks_gen_L_v2.1)

AnandRaj-R, 2022-03-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I connect Zonestar 2004 keypad lcd (Anet A8 lcd) to EXP1 on Mks Gen-L v2.1(Ramps 1.6+) lcd is working perfectly
but keypad buttons are not working.
My Pin config: -   (for Zonestar keypad Lcd)

      #define LCD_PINS_RS                     25     //Default Pin 64
      #define LCD_PINS_ENABLE                 16     //Default Pin44
      #define LCD_PINS_D4                     27     //Default Pin63
      #define LCD_PINS_D5                     23     //Default Pin40
      #define LCD_PINS_D6                     17     //Default Pin42
      #define LCD_PINS_D7                     37     //Default Pin65
      #define ADC_KEYPAD_PIN                  57 (A3) or 58 (A4)  //Default Pin 12

I cannot connect the keypad pin to default ADC KEYPAD PIN 12 because the board not have a physical pin 12 so i changed the default ADC KEYPAD PIN 12 to 57 or 58 then i got below compiler error.

(i cannot understand this pin 12 is digital or analog, if it is digital, keypad won't work because we need an adc pin or if it is analog pin, we need to change this pin 12 to 66)

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.3
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.5.1
|   |-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0

...
...
...

_In file included from Marlin\src\module\../inc/../HAL/./AVR/HAL.h:22:0,
                 from Marlin\src\module\../inc/../HAL/HAL.h:30,
                 from Marlin\src\module\../inc/MarlinConfig.h:31,
                 from Marlin\src\module\../MarlinCore.h:24,
                 from Marlin\src\module\temperature.cpp:30:
Marlin\src\module\temperature.cpp: In member function 'void Temperature::init()':
Marlin\src\module\../inc/../HAL/./AVR/../shared/Marduino.h:42:24: warning: left shift count >= width of type [-Wshift-count-overflow]
 #define _BV(b) (1 << (b))
                        ^
Marlin\src\module\../inc/../HAL/./AVR/../shared/Marduino.h:44:26: note: in expansion of macro '_BV'
   #define SBI(A,B) (A |= _BV(B))
                          ^~~
Marlin\src\module\../inc/../HAL/./AVR/HAL.h:171:51: note: in expansion of macro 'SBI'
   #define HAL_ANALOG_SELECT(ind) do{ if (ind < 8) SBI(DIDR0, ind); else SBI(DIDR2, ind & 0x07); }while(0)
                                                   ^~~
Marlin\src\module\temperature.cpp:2389:5: note: in expansion of macro 'HAL_ANALOG_SELECT'
     HAL_ANALOG_SELECT(ADC_KEYPAD_PIN);
     ^~~~~~~~~~~~~~~~~_

### Bug Timeline

new bug

### Version of Marlin Firmware

Marlin 2.0.9.3

### Printer model

Anet A8

### Electronics

Mks Gen-L v2.1(atmega2560)

### Additional information & file uploads

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/8342251/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/8342253/Configuration_adv.h.txt)
[pins_RAMPS.h.txt](https://github.com/MarlinFirmware/Marlin/files/8342255/pins_RAMPS.h.txt)
[MKS GEN_L V2.1_001 PIN.pdf](https://github.com/MarlinFirmware/Marlin/files/8342263/MKS.GEN_L.V2.1_001.PIN.pdf)
[image]
[image]

ellensp, 2022-03-24:
`#define ADC_KEYPAD_PIN                  57 (A3) or 58 (A4)  //Default Pin 12 
`
is not a valid define

Analog pins are 0-15  

so if you want A3 use #define ADC_KEYPAD_PIN 3
so if you want A4 use #define ADC_KEYPAD_PIN 4

ellensp, 2022-03-24:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

AnandRaj-R, 2022-03-24:
I understood thanks to all.

AnandRaj-R, 2022-03-24:
After changing ADC_KEYPAD_PIN 57 to 3(A3) now Keypad works perfectly, thanks for quick response.
[image]