MarlinFirmware/Marlin #23942: [FR] Invert FREEZE_FEATURE polarity

KillerBug, 2022-03-24:
### Describe the feature you want

Currently the FREEZE_FEATURE only works with normally-open switches.  Most emergency stop switches are normally-closed.  It would be nice if there was something that could make it activate when high instead of when low.

ellensp, 2022-03-25:
This should add it 
```DIFF
+++ b/Marlin/Configuration_adv.h
@@ -4113,6 +4113,7 @@
 //#define FREEZE_FEATURE
 #if ENABLED(FREEZE_FEATURE)
   //#define FREEZE_PIN 41   // Override the default (KILL) pin here
+  //#define INVERT_FREEZE_PIN 
 #endif
 
 /**
diff --git a/Marlin/src/MarlinCore.cpp b/Marlin/src/MarlinCore.cpp
index 4117d3f382..0e2c9b6efa 100644
--- a/Marlin/src/MarlinCore.cpp
+++ b/Marlin/src/MarlinCore.cpp
@@ -476,7 +476,7 @@ inline void manage_inactivity(const bool no_stepper_sleep=false) {
   #endif
 
   #if HAS_FREEZE_PIN
-    Stepper::frozen = !READ(FREEZE_PIN);
+    Stepper::frozen = !READ(FREEZE_PIN)^ ENABLED(INVERT_FREEZE_PIN;
   #endif
 
   #if HAS_HOME

```

ellensp, 2022-03-25:
This should work
```DIFF
diff --git a/Marlin/Configuration_adv.h b/Marlin/Configuration_adv.h
index 0a18e95ebc..1577f144fa 100644
--- a/Marlin/Configuration_adv.h
+++ b/Marlin/Configuration_adv.h
@@ -4113,6 +4113,7 @@
 //#define FREEZE_FEATURE
 #if ENABLED(FREEZE_FEATURE)
   //#define FREEZE_PIN 41   // Override the default (KILL) pin here
+  //#define INVERT_FREEZE_PIN 
 #endif
 
 /**
diff --git a/Marlin/src/MarlinCore.cpp b/Marlin/src/MarlinCore.cpp
index 4117d3f382..2dcc6bfdae 100644
--- a/Marlin/src/MarlinCore.cpp
+++ b/Marlin/src/MarlinCore.cpp
@@ -476,7 +476,7 @@ inline void manage_inactivity(const bool no_stepper_sleep=false) {
   #endif
 
   #if HAS_FREEZE_PIN
-    Stepper::frozen = !READ(FREEZE_PIN);
+    Stepper::frozen = READ(FREEZE_PIN) ^ DISABLED(INVERT_FREEZE_PIN);
   #endif
 
   #if HAS_HOME
```

KillerBug, 2022-03-25:
> 

YES!  That works.  Gets me going but could it be built-in so modding MarlinCore.cpp isn't required?

closed 2022-03-25 by ellensp