MarlinFirmware/Marlin #23947: [BUG] Octopus and Inductive sensor needs ENDSTOPPULLDOWN_ZMIN_PROBE but it's now blocked by the insanity check in error

Aurawin, 2022-03-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

  When compiling marlin I now get an error 

Marlin\src\HAL\STM32\../../inc/SanityCheck.h:794:4: error: #error "PULLDOWN pin mode is not available on the selected board."

#define ENDSTOPPULLDOWN_ZMIN_PROBE 

I've been running this config for a year with no issues.  Wondering if this was done in error.

### Bug Timeline

Current bugfixes-2.0.x

### Expected behavior

#define ENDSTOPPULLDOWN_ZMIN_PROBE 

should compile for Octopus V1.1

### Actual behavior

Marlin\src\HAL\STM32\../../inc/SanityCheck.h:794:4: error: #error "PULLDOWN pin mode is not available on the selected board."
  794 |   #error "PULLDOWN pin mode is not available on the selected board."

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Tronxy X5S

### Electronics

BigTreeTech Octopus V1_1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

Aurawin, 2022-03-26:
It should be noted that homing cannot occur with this option.

thinkyhead, 2022-03-26:
Including your configuration files allows us to instantly test the build and work on a fitting solution.

thinkyhead, 2022-03-26:
Yep, it looks like a few more boards have a distinct pulldown options, not just the LPC176x boards as first applied. I'll have the sanity check patched shortly. Thanks for the report!

closed 2022-03-26 by thinkyhead