MarlinFirmware/Marlin #23951: [BUG] Make fails with "Makefile:135: *** missing separator.  Stop."

drf5n, 2022-03-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

On branch bugfix-2.0.x a plan make fails:

```
% git checkout bugfix-2.0.x
Already on 'bugfix-2.0.x'
Your branch is up to date with 'origin/bugfix-2.0.x'.
drf@V50833 Marlin % git pull                 
X11 forwarding request failed on channel 0
Already up to date.
% pwd
/Users/drf/Work/Marlin/Marlin
% make
Makefile:135: *** missing separator.  Stop.
% 

```

Line 135 is this:

https://github.com/MarlinFirmware/Marlin/blob/4a54f8469e66a4f59c6284646885545d52f6e164/Marlin/Makefile#L135

If you replace the two spaces with a tab, you get a different error:

```
% make
Makefile:135: *** commands commence before first target.  Stop.
```
If you comment it out, it works.

If you replace it with $(warning .... ) it works

### Bug Timeline

Not sure

### Expected behavior

I expected the makefile to compile.

### Actual behavior

The make process failed with an error

### Steps to Reproduce

% git checkout bugfix-2.0.x
Already on 'bugfix-2.0.x'
Your branch is up to date with 'origin/bugfix-2.0.x'.
drf@V50833 Marlin % git pull                 
X11 forwarding request failed on channel 0
Already up to date.
% pwd
/Users/drf/Work/Marlin/Marlin
% make
Makefile:135: *** missing separator.  Stop.
% 

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

ramps 1.4

### Electronics

ramps 1.4

### Add-ons

--

### Additional information & file uploads

I'm on MacOS Big Sur 11.6.5 using /usr/bin/make:

```
% uname -a
Darwin V50833.local 20.6.0 Darwin Kernel Version 20.6.0: Tue Feb 22 21:10:41 PST 2022; root:xnu-7195.141.26~1/RELEASE_X86_64 x86_64
% which make
/usr/bin/make
% make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0
% 

```

Again, replacing the `@echo ...` line with `$(warning ...)` solves the problem.

ellensp, 2022-03-27:
# Detailed instructions for using the makefile:
#
#  1. Modify the line containing "ARDUINO_INSTALL_DIR" to point to the directory that
#     contains the Arduino installation (for example, under macOS, this
#     might be /Applications/Arduino.app/Contents/Resources/Java).
#
#  2. Modify the line containing "UPLOAD_PORT" to refer to the filename
#     representing the USB or serial connection to your Arduino board
#     (e.g. UPLOAD_PORT = /dev/tty.USB0).  If the exact name of this file
#     changes, you can use * as a wild card (e.g. UPLOAD_PORT = /dev/tty.usb*).
#
#  3. Set the line containing "MCU" to match your board's processor. Set
#     "PROG_MCU" as the AVR part name corresponding to "MCU". You can use the
#     following command to get a list of correspondences: `avrdude -c alf -p x`
#     Older boards are atmega8 based, newer ones like Arduino Mini, Bluetooth
#     or Diecimila have the atmega168.  If you're using a LilyPad Arduino,
#     change F_CPU to 8000000. If you are using Gen7 electronics, you
#     probably need to use 20000000. Either way, you must regenerate
#     the speed lookup table with create_speed_lookuptable.py.
#
#  4. Type "make" and press enter to compile/verify your program.
#
#  5. Type "make upload", reset your Arduino board, and press enter to
#     upload your program to the Arduino board.
#
# Note that all settings at the top of this file can be overridden from
# the command line with, for example, "make HARDWARE_MOTHERBOARD=71"
#
# To compile for RAMPS (atmega2560) with Arduino 1.6.9 at root/arduino you would use...
#
#   make ARDUINO_VERSION=10609 AVR_TOOLS_PATH=/root/arduino/hardware/tools/avr/bin/ \
#   HARDWARE_MOTHERBOARD=1200 ARDUINO_INSTALL_DIR=/root/arduino
#
# To compile and upload simply add "upload" to the end of the line...
#
#   make ARDUINO_VERSION=10609 AVR_TOOLS_PATH=/root/arduino/hardware/tools/avr/bin/ \
#   HARDWARE_MOTHERBOARD=1200 ARDUINO_INSTALL_DIR=/root/

drf5n, 2022-03-27:
The conditional wrapped around https://github.com/MarlinFirmware/Marlin/blob/4a54f8469e66a4f59c6284646885545d52f6e164/Marlin/Makefile#L134 makes this error only occur for the sub-40901 versions of C it is aimed at:

```
CC_MAJ:=$(shell $(CC) -dM -E - < /dev/null | grep __GNUC__ | cut -f3 -d\ )
CC_MIN:=$(shell $(CC) -dM -E - < /dev/null | grep __GNUC_MINOR__ | cut -f3 -d\ )
CC_PATCHLEVEL:=$(shell $(CC) -dM -E - < /dev/null | grep __GNUC_PATCHLEVEL__ | cut -f3 -d\ )
CC_VER:=$(shell echo $$(( $(CC_MAJ) * 10000 + $(CC_MIN) * 100 + $(CC_PATCHLEVEL) )))
ifeq ($(shell test $(CC_VER) -lt 40901 && echo 1),1)
  @echo This version of GCC is likely broken. Enabling relocation workaround.
  RELOC_WORKAROUND = 1
endif
```

If I change the conditional to: 

```
ifeq ($(shell test $(CC_VER) -lt 30901 && echo 1),1)
```

... then make doesn't choke on the @echo line

My version of CC is:

```
% gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 13.0.0 (clang-1300.0.29.30)
Target: x86_64-apple-darwin20.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
If I use a line like 

```
  $(warning @echo This version $(CC_VER) of GCC is likely broken. Enabling relocation workaround.)
```
I get:

```
% time make ARDUINO_VERSION=10609 UPLOAD_PORT=/dev/cu.usbserial-143330  -j 8 all
Makefile:136: @echo This version 40201 of GCC is likely broken. Enabling relocation workaround.
  CC    /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring.c
  CC    /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_analog.c
  CC    /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_digital.c
  CC    /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_shift.c
  CC    /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c
  CC    /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/hooks.c
  CC    /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_pulse.c
  CXX   /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WMath.cpp
...

```
...and the compilation works.

The conditional-protected line #134 is poorly formatted for a Makefile, because the command isn't preceded by the required  tab character.  If you replace the spaces with a tab character, it fails because the command is early in the Makefile parsing and comes after no Makefile targets.   

One workaround would be to update the compiler to avoid this broken exception code.

One workaround would be to change the character before the `@echo` command and add a dummy target before it, but it would confuse the normal processing with the new default target.

A better workaround would be to avoid calling the parent shell's echo command use Make's built-in warning mechanism with a line like:

```
$(warning This GCC version $(CC_VER) is likely broken. Enabling relocation workaround.)
``` 

I'll rig up a pull request.

thisiskeithb, 2022-03-27:
Closing since you’ve opened a PR.

closed 2022-03-27 by thisiskeithb