MarlinFirmware/Marlin #23952: [BUG] Compile issue bugfix2.0.x - STM32 fast_pwm.cpp HT not declared

LightningShark25, 2022-03-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Compile issue bugfix2.0.x - STM32 fast_pwm.cpp HT not declared in scope

env: BIGTREE_SKR_2

Successful compile before refreshing bugfix2.0.x to local branch.
Will not compile due to following.  Hopefully this is the right place to detail.

Will compile with default bugfix2.0.x board configs etc.  But will not compile for BIGTREE_SKR_2 board.

Compiling .pio\build\BIGTREE_SKR_2\src\src\HAL\STM32\tft\tft_fsmc.cpp.o
Marlin\src\HAL\STM32\fast_pwm.cpp: In static member function 'static void MarlinHAL::set_pwm_duty(pin_t, uint16_t, uint16_t, bool)':
Marlin\src\HAL\STM32\fast_pwm.cpp:60:3: error: 'HT' was not declared in this scope
   60 |   HT = (HardwareTimer *)(HardwareTimer_Handle[index]->__this);
      |   ^~
Marlin\src\HAL\STM32\fast_pwm.cpp:60:52: error: invalid types 'timerObj_t* [14][char*(const char*, int)]' for array subscript
   60 |   HT = (HardwareTimer *)(HardwareTimer_Handle[index]->__this);
      |                                                    ^
In file included from C:\Users\Kevin\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32/PinNames.h:4,
                 from C:\Users\Kevin\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32/pinmap.h:22,
                 from C:\Users\Kevin\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32/PeripheralPins.h:34,
                 from C:\Users\Kevin\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32/analog.h:44,
                 from C:\Users\Kevin\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino/board.h:8,
                 from C:\Users\Kevin\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino/wiring.h:41,
                 from C:\Users\Kevin\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino/Arduino.h:36,
                 from Marlin\src\HAL\STM32\../../inc/../HAL/./STM32/../shared/Marduino.h:36,
                 from Marlin\src\HAL\STM32\../../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin\src\HAL\STM32\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32\fast_pwm.cpp:27:
Marlin\src\HAL\STM32\fast_pwm.cpp:61:54: error: 'pin_name' was not declared in this scope; did you mean 'pin_pinName'?
   61 |   uint32_t channel = STM_PIN_CHANNEL(pinmap_function(pin_name, PinMap_PWM));
      |                                                      ^~~~~~~~
C:\Users\Kevin\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\cores\arduino\stm32/PinNamesTypes.h:85:40: note: in definition of macro 'STM_PIN_CHANNEL'
   85 | #define STM_PIN_CHANNEL(X)          (((X) >> STM_PIN_CHAN_SHIFT) & STM_PIN_CHAN_MASK)
      |                                        ^
Marlin\src\HAL\STM32\fast_pwm.cpp:62:3: error: 'previousMode' was not declared in this scope
   62 |   previousMode = HT->getMode(channel);
      |   ^~~~~~~~~~~~
Marlin\src\HAL\STM32\fast_pwm.cpp:67:7: error: 'needs_freq' was not declared in this scope
   67 |   if (needs_freq) {
      |       ^~~~~~~~~~
compilation terminated due to -fmax-errors=5.
Compiling .pio\build\BIGTREE_SKR_2\src\src\HAL\STM32\tft\tft_ltdc.cpp.o
*** [.pio\build\BIGTREE_SKR_2\src\src\HAL\STM32\fast_pwm.cpp.o] Error 1

### Bug Timeline

New - after latest merge bugfix2.0.x

### Expected behavior

Compile

### Actual behavior

Fail as above

### Steps to Reproduce

Configure autobuild marlin for BIGTREE_SKR_2 and compile

### Version of Marlin Firmware

Bugfix2.0.x latest 27/3/2022

LightningShark25, 2022-03-27:
Please advise if this is the not method for logging compile bugs - I'm not a software developer, but have small level of code compiling experience with Marlin

ellensp, 2022-03-27:
Cannot reproduce.

Please provide Configuration files

LightningShark25, 2022-03-27:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/8357358/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/8357359/Configuration_adv.txt)

LightningShark25, 2022-03-27:
Files above are .h with extension renamed to meet github upload rules.

Not sure how many other files I've customised, mostly around getting the pin mapping and standard ender 3 v2 display to work with the SKR board.    I suspect you might run into trouble with that, but see how you go

ellensp, 2022-03-27:
Provided configs are for RAMPs
  #define MOTHERBOARD BOARD_RAMPS_14_EFB

LightningShark25, 2022-03-27:
Apologies - not sure what I've done :( - will recheck and upload right ones soon

LightningShark25, 2022-03-27:
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/8357491/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/8357492/Configuration.txt)

Here are correct configs - I copied from the wrong repo sorry (which I will now delete)

ellensp, 2022-03-27:
Dont get any error like that. 

I do however get 

#error "INDIVIDUAL_AXIS_HOMING_SUBMENU is required with ProUI."
#error "LCD_SET_PROGRESS_MANUALLY is required with ProUI."
#error "STATUS_MESSAGE_SCROLLING is required with ProUI."
#error "BAUD_RATE_GCODE is required with ProUI."
#error "SOUND_MENU_ITEM is required with ProUI."
#error "PRINTCOUNTER is required with ProUI."

LightningShark25, 2022-03-27:
Thanks for looking at it.  

I was expecting some complaints about the UI/ LCD but didn't get that far in the compile.
I previously had it compiling with extended UI but I did see when I was resolving some merge conflicts that it may now be superceded/replaced by PROUI.  

Not sure where it has all gone wrong - if I shift the head back prior to merge I can still compile...

ellensp, 2022-03-27:
Seems to be some sort of user error. closing.

closed 2022-03-27 by ellensp

LightningShark25, 2022-03-27:
FWIW I ditched my repo and took my configs across to a new clone of the bugfix repo.  
Something got corrupted in there badly.  First run of the new repo and got through all good.
@ellensp  appreciate your time - you made the decision to do this easy.
Now to recustomise...