MarlinFirmware/Marlin #23953: Probe fail! @ high temprature

sinole1987, 2022-03-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have SKR 1.4T and I use a PINDA 2.0 probe on my bed leveling.
I didn't have this issue in older marlin 2.0 but I recently updated to 2.0.9 and  I can calibrate my probe and probe the bed at high temperature. I get a "Probe fail!" massage  when drift goes to high.
same issue was solved by a user before [[BUG] Probing Failed](https://github.com/MarlinFirmware/Marlin/issues/20442), but I am not a programmer and I cant understand how person solved the problem. I went as far as changing my pinda probe 2 times. I don't think this is a hardware issue. help me understand why this error occurs and how we cant fix it.

### Bug Timeline

new, after 2.0.9

### Version of Marlin Firmware

9.0.9.3

### Bed Leveling

ABL Bilinear mesh

sinole1987, 2022-04-21:
ye I saw that in previous logs and tried it. It didn't work. before I used to get this error only there was a piece of filament or debris under the flex plate. i thought its a built in feature in marlin to detect very uneven bed. my printer bed is quit flat. I thought maybe there is a set slope under which that error pops and now its too aggressive. is that a thing? and if so can I change that?  I doubt that is the case since this error only happens with temperature change, actually I "0" my temperature compensation values and didn't get  the error. but it worth to try and change that slope if its a thing.

thinkyhead, 2022-04-26:
Are you using the `PROBE_TRIGGERED_WHEN_STOWED_TEST` option?

sinole1987, 2022-04-27:
question!? if I copy configuration.h from current firmware to bugfix will it work? setting that up is a pain. my printer is costume made.

thisiskeithb, 2022-04-27:
> question!? if I copy configuration.h from current firmware to bugfix will it work?

Not likely if you didn’t test bugfix before opening this issue like the form asked 🙂

Use your favorite diff tool (Meld, Beyond Compare, etc.) to bring your config up to date with bugfix and run through the test procedure that thinkyhead lists above.

sinole1987, 2022-04-27:
I tested bugfix before, I don't have the files. that's why I asked. believe it or not I sit hours and config my printer every time new version of marlin comes. my programing skills are nonexistent. ill check those programs, thank you. The reason I rolled back to 2.0.9 was that I though it might be more stable.

sinole1987, 2022-04-28:
I did it, I coppied everything that Pronterface held. but i think last few lines is where error occurs.
[bug.txt](https://github.com/MarlinFirmware/Marlin/files/8578996/bug.txt)

sinole1987, 2022-04-30:
> Let's collect more data…
> 
> * Download Marlin `bugfix-2.0.x` to test with the latest code.
> * Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
> * Connect to your printer from host software such as Cura, Printrun or Repetier Host.
> * Send `M502` and `M500` to ensure your Configurations are applied.
> * Issue the command `M111 S247` to enable maximum logging.
> * Perform a `G28` to do your standard homing procedure.
> * Do a `G29` to probe the bed. This will also enable bed leveling.
> * Copy the log output into a **.TXT file** and attach it to your next reply.
> 
> Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

I just wanted to let guys know the errors got more frequent since I installed bugfix, let me know if you want me to collect more data.

sinole1987, 2022-05-30:
I think I have solved the issue. My print bed was very flat with maximum of 0.1mm differential height so i was afforded to do a very aggressive bed leveling with #define Z_CLEARANCE_MULTI_PROBE  of 0.4 mm. after checking the mesh again i found out that the thick aluminum bed has wrapped a little and now there is more than 0.3mm of differential height. I think when it heats up it gets to 0.4 and triggers the early trigger message. I increased the probing dive to 0.8mm for now. In the meantime I apologize for wasting your valuable time. I will close this issue and report in case more testing had different results. fort now it seems to be solved.

closed 2022-05-30 by sinole1987