MarlinFirmware/Marlin #23955: After update to 2.9.3 stuck on black marlin boot screen

SamppaD78, 2022-03-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After Marlin update printer fails to reboot to home screen. Main board accept firmware as file gets .cur extension but will not boot past Marlin boot logo.I have built a firmwiare with  a STM32F103RC_btt 256K as always, I have re flash multiple times,recompiled firmware and tested bug,fix. Tried a fix  that was on github page to  modify Marlin  hal.h  and marlin.cpp files to resolve this problem but it ended with a errors  after updating files. At the moment no solution to this problem. It seems like every time i manage to fix bugs I come across a new one with latest marlin updates.
So printer is not operational,no response in marlin mode ,I can access SKR screen touch mode  but no connection to main board error is displayed. This is actually first time when this issue happens,normally there wasnt issues with updating mainboard.

VScode is the latest all updates installed.
Marlin is latest version
Any help to resolve this will be highly appreciated

### Bug Timeline

New

### Expected behavior

After firmwire is flashed normal reboot to home screen is expected.

### Actual behavior

After firmwire is flashed printer mainboard is stuck on black screen  with marlin logo

### Steps to Reproduce

Flash the latest Marlin 
Recompile for SKR mini E3 2.0 Ender 3 Pro
Wait for printer to install and reboot.
Flashing fails and main board is frozen,it will accept new firmware but will not reboot to home screen.

### Version of Marlin Firmware

2.9.3

### Printer model

Ender 3 Pro

### Electronics

SKR mini E3 2.0

### Add-ons

no

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

SamppaD78, 2022-03-27:
Update on this issue. Tried to flash a working backup firmware and noticed that file does not get a .cur extension. So it looks like flashing via mainboard usb is not working anymore.

SamppaD78, 2022-03-27:
I can confirm that flashing a new firmware via Sd card slot or usb port is not working. Any options to reset board?

thisiskeithb, 2022-03-27:
Try a new SD card? Your bootloader handles flashing, not Marlin.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-03-27 by thisiskeithb

SamppaD78, 2022-03-27:
Yes. Multiple  sd cards  that i tested and same issue. Turned off printer for some time and  flashed a new firmware to the printer. Noticed that this time boot-loader will update firmware leaving .cur mark but printer  will not reboot to home screen. Only black screen with  marlin mode  is visible on the screen. I can still enter a touch screen but no connection to the board is displayed. main-board has a propper  24V ,3.3V and 5V present. It looks like bootloader is stuck after update?

SamppaD78, 2022-03-27:
Update:
Disconnected  Bltouch,flashed BTT original firmware no help. Try to reset main board via reset switch,nothing happens. Main board is powered all voltages are present,also solid red and solid green led are on.