MarlinFirmware/Marlin #23956: [BUG] Mightyboard Rev E pin definitions incorrect

jamespearson04, 2022-03-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The pins for extruder 2 heater and fan are swapped. This causes the heater to behave as the fan, switching on once the heating from extruder 1 causes extruder 2 to reach 50C - heating until the printer is switched off. (Significant fire risk if unattended)

The MOSFET pin definitions are currently (from line 150 in pins_MIGHTYBOARD_REV_E.h):
#define MOSFET_1_PIN                           6  // Plug EX1 Pin 1-2 -> PH3 #15 -> Logical 06
#define MOSFET_2_PIN                           7  // Plug EX1 Pin 3-4 -> PH4 #16 -> Logical 07
#define MOSFET_3_PIN                          12  // Plug EX2 1-2 -> PB5 #25 -> Logical 12
#define MOSFET_4_PIN                          11  // Plug EX2 3-4 -> PB6 #24 -> Logical 11
#define MOSFET_5_PIN                          45  // Plug HBD 1-2 -> PL4 #39 -> Logical 45
#define MOSFET_6_PIN                          13  // Plug Extra 1-2 -> PL5 #40 -> Logical 44 (FET not soldered in all boards)

They should be:
#define MOSFET_1_PIN                           6  // Plug EX1 Pin 1-2 -> PH3 #15 -> Logical 06
#define MOSFET_2_PIN                           7  // Plug EX1 Pin 3-4 -> PH4 #16 -> Logical 07
#define MOSFET_3_PIN                          **11**  // Plug EX2 1-2 -> PB5 #25 -> Logical 12
#define MOSFET_4_PIN                          **12**  // Plug EX2 3-4 -> PB6 #24 -> Logical 11
#define MOSFET_5_PIN                          45  // Plug HBD 1-2 -> PL4 #39 -> Logical 45
#define MOSFET_6_PIN                          13  // Plug Extra 1-2 -> PL5 #40 -> Logical 44 (FET not soldered in all boards)

### Bug Timeline

Unknown, new user

### Expected behavior

Fan switches on when extruder 2 reaches 50C.
Heater switches on as required to heat extruder 2.

### Actual behavior

Heater switches on when extruder 2 reaches 50C - Continues to heat until the machine is switched off.
Fan switches on when heater is supposed to start heating - warning alarm triggers after a few seconds.

### Steps to Reproduce

1. Flash onto Mightyboard rev e
2. Attempt to heat extruder 2

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

CTC Bizer

### Electronics

Stock electronics + Bltouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

jamespearson04, 2022-03-31:
Since found that MOSFET_6 is also wrong, should be defined as **44**

`#define MOSFET_1_PIN                           6  // Plug EX1 Pin 1-2 -> PH3 #15 -> Logical 06
#define MOSFET_2_PIN                           7  // Plug EX1 Pin 3-4 -> PH4 #16 -> Logical 07
#define MOSFET_3_PIN                          11  // Plug EX2 1-2 -> PB5 #24 -> Logical 11
#define MOSFET_4_PIN                          12  // Plug EX2 3-4 -> PB6 #25 -> Logical 12
#define MOSFET_5_PIN                          45  // Plug HBD 1-2 -> PL4 #39 -> Logical 45
#define MOSFET_6_PIN                          44  // Plug Extra 1-2 -> PL5 #40 -> Logical 44 (FET not soldered in all boards)

//
// Heaters / Fans (24V)
//

#define HEATER_0_PIN                MOSFET_1_PIN // EX1
#define HEATER_1_PIN                MOSFET_3_PIN // EX2
#define HEATER_BED_PIN              MOSFET_5_PIN // HBP

#ifdef NUM_M106_FANS
  #define FAN_PIN                     MOSFET_6_PIN //Print fan
#endif`

Since this section has attention, may I also suggest adding the last 3 lines shown above to the heater/fan definitions, as MOSFET 6 is the typical route for adding a cooling fan on this board?

grauerfuchs, 2022-03-31:
Good catch. Again, you're correct on the pin. I should have seen that; it was right there with the wrong assignment even though the description beside it was correct.

As for setting the FAN_PIN, this may or may not be desirable. As you surmised, the MOSFET assignments came from the RAMPS-style auto-assignment. This was intentional so the different boards used in a generic fashion more or less behave the same way. However, it's not the correct assignment for the Replicator, the FFCP, or any of the clones. In that, the Mightyboard is a bit of an odd duck because it was built for specific machines and not for general usage.

Directly setting the pin's function outside of the unified generic logic may be better handled in the configuration as was done in the example config as was done for FFCP.

@thinkyhead, do you want to weigh in on this one?

thisiskeithb, 2022-03-31:
Closing since a PR was opened: #23976

closed 2022-03-31 by thisiskeithb