MarlinFirmware/Marlin #23958: [BUG] ADAPTIVE_STEP_SMOOTHING breaks Bed Leveling on SKR Mini E3 v3

D3strukt0r, 2022-03-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have (in my opinion) correctly installed the mainboard, but I'm happy to provide a screenshot if you want to.
When I want to do anything with this mainboard I get errors. This has never happened to my v2 board. Right now, for example, I'm unable to do the bed levelling (G29 T (Bilinear)). This will mostly happen when I don't just do Bed Leveling on but when I also heat up the bed. On the screen I get the error:

M112 Shutdown:
PRINTER HALTED
Please reset

In OctoPrint I can see:

Recv: ok
Send: G29 T
[...]
Recv: [//action:notification](https://action:notification) Probing Failed
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

You can find my current Marlin files (which I used for Marlin v2.0.9.3 and bugfix-2.x), as well as videos and screenshots of it happening in the following Dropbox folder:
https://www.dropbox.com/sh/7nhgjux8ageesjb/AADQE8aKNhegGgSuZD0KEAXka?dl=0

The message above is what I sent to BTT support, but without an answer, since then I've brought the config to a somewhat stable state through trial and error. I seem to not be getting crashes with it and the Bed Leveling seems to work (even though a bit noisy).

[image]

The moment I turn on `ADAPTIVE_STEP_SMOOTHING` though, the result is either that Bed Leveling looks like this:

[image]

or it crashes like shown in the videos in my Dropbox folder

### Bug Timeline

Ever since using the new board (SKR Mini E3 v3)

### Expected behavior

It works

### Actual behavior

It doesn't work lol

### Steps to Reproduce

1. I guess build Marlin using my config
2. Do the bed levelling (`G28` and `G29 T`)
3. See it work
4. Enable `ADAPTIVE_STEP_SMOOTHING` and build again
5. Do the bed levelling
6. See it fail

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Mar 27 2022 20:04:20)

### Printer model

Creality Ender 3

### Electronics

SKR Mini E3 v3

### Add-ons

BL Touch Smart v3.1, Rollout Sensor, Stock Display

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

https://www.dropbox.com/sh/7nhgjux8ageesjb/AADQE8aKNhegGgSuZD0KEAXka?dl=0

CRCinAU, 2022-03-29:
On a hunch, can you build with M48 (Z_MIN_PROBE_REPEATABILITY_TEST) enabled, and then run:

`M48 V4 P20`

For reference, here are my results using the same board:
```
Recv: 1 of 20: z: 0.004 Mean: 0.003750 Sigma: 0.000000 Min: 0.004 Max: 0.004 Range: 0.000
Recv: 2 of 20: z: 0.005 Mean: 0.004375 Sigma: 0.000625 Min: 0.004 Max: 0.005 Range: 0.001
Recv: 3 of 20: z: -0.197 Mean: -0.062917 Sigma: 0.095166 Min: -0.197 Max: 0.005 Range: 0.202
Recv: 4 of 20: z: 0.005 Mean: -0.045938 Sigma: 0.087506 Min: -0.197 Max: 0.005 Range: 0.202
Recv: 5 of 20: z: -0.329 Mean: -0.102500 Sigma: 0.137561 Min: -0.329 Max: 0.005 Range: 0.334
Recv: 6 of 20: z: 0.005 Mean: -0.084583 Sigma: 0.131812 Min: -0.329 Max: 0.005 Range: 0.334
Recv: 7 of 20: z: -0.264 Mean: -0.110179 Sigma: 0.137197 Min: -0.329 Max: 0.005 Range: 0.334
Recv: 8 of 20: z: 0.007 Mean: -0.095469 Sigma: 0.134107 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 9 of 20: z: -0.296 Mean: -0.117778 Sigma: 0.141308 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 10 of 20: z: 0.004 Mean: -0.105625 Sigma: 0.138926 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 11 of 20: z: 0.005 Mean: -0.095568 Sigma: 0.136225 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 12 of 20: z: -0.139 Mean: -0.099167 Sigma: 0.130970 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 13 of 20: z: 0.006 Mean: -0.091058 Sigma: 0.128929 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 14 of 20: z: 0.006 Mean: -0.084107 Sigma: 0.126742 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 15 of 20: z: 0.004 Mean: -0.078250 Sigma: 0.124390 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 16 of 20: z: 0.004 Mean: -0.073125 Sigma: 0.122065 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 17 of 20: z: 0.002 Mean: -0.068676 Sigma: 0.119750 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 18 of 20: z: 0.004 Mean: -0.064653 Sigma: 0.117552 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 19 of 20: z: 0.002 Mean: -0.061118 Sigma: 0.115395 Min: -0.329 Max: 0.007 Range: 0.336
Recv: 20 of 20: z: 0.001 Mean: -0.058000 Sigma: 0.113292 Min: -0.329 Max: 0.007 Range: 0.336
Recv: Finished!
Recv: Mean: -0.058000 Min: -0.329 Max: 0.007 Range: 0.336
Recv: Standard Deviation: 0.113292
```

I'm curious to figure out if yours is similar - or if theres something else going on.... My results above seem to show 15 really good results (+/- 0.007mm) - but the other 5 results are waaaaay out... I'm just trying to see if what I'm chasing is related...

thisiskeithb, 2022-03-29:
The MCU on this board is not as powerful as previous generations, so `ADAPTIVE_STEP_SMOOTHING` should probably be disabled.

CRCinAU, 2022-03-30:
@D3strukt0r - Check out the latest example configurations, turn off `ADAPTIVE_STEP_SMOOTHING` and test this all again.

Let us know if things improve.

D3strukt0r, 2022-03-30:
Thanks for the quick help, will try it out and report in the next free moment I have @CRCinAU @thisiskeithb

thisiskeithb, 2022-04-03:
`ADAPTIVE_STEP_SMOOTHING` [is now disabled](https://github.com/MarlinFirmware/Configurations/blob/5147e6b2ad822b676f19c819013fe7debc72aaca/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%203.0/Configuration_adv.h#L1197-L1203) for the Ender-3 w/ BigTreeTech SKR Mini E3 3.0 config.

closed 2022-04-03 by thisiskeithb

D3strukt0r, 2022-04-03:
@CRCinAU
I tried it out again on the current latest commit c4873a64ecd54dc0bbccb1eaeecd1b5ae2f6aef5 (04.04.2022)
... without `ADAPTIVE_STEP_SMOOTHING `
```
Send: M48 V4 P20
Recv: M48 Z-Probe Repeatability Test
Recv: Positioning the probe...
Recv: Bed X: 117.00 Y: 109.00 Z: 0.00
Recv: 1 of 20: z: 0.013 Mean: 0.012500 Sigma: 0.000000 Min: 0.013 Max: 0.013 Range: 0.000
Recv: 2 of 20: z: 0.013 Mean: 0.012500 Sigma: 0.000000 Min: 0.013 Max: 0.013 Range: 0.000
Recv: 3 of 20: z: 0.013 Mean: 0.012500 Sigma: 0.000000 Min: 0.013 Max: 0.013 Range: 0.000
Recv: 4 of 20: z: 0.015 Mean: 0.013125 Sigma: 0.001082 Min: 0.013 Max: 0.015 Range: 0.002
Recv: 5 of 20: z: 0.017 Mean: 0.014000 Sigma: 0.002000 Min: 0.013 Max: 0.017 Range: 0.005
Recv: 6 of 20: z: 0.020 Mean: 0.015000 Sigma: 0.002887 Min: 0.013 Max: 0.020 Range: 0.007
Recv: 7 of 20: z: 0.023 Mean: 0.016071 Sigma: 0.003746 Min: 0.013 Max: 0.023 Range: 0.010
Recv: 8 of 20: z: 0.023 Mean: 0.016875 Sigma: 0.004098 Min: 0.013 Max: 0.023 Range: 0.010
Recv: 9 of 20: z: 0.025 Mean: 0.017778 Sigma: 0.004631 Min: 0.013 Max: 0.025 Range: 0.012
Recv: 10 of 20: z: 0.025 Mean: 0.018500 Sigma: 0.004899 Min: 0.013 Max: 0.025 Range: 0.012
Recv: 11 of 20: z: 0.027 Mean: 0.019318 Sigma: 0.005340 Min: 0.013 Max: 0.027 Range: 0.015
Recv: 12 of 20: z: 0.027 Mean: 0.020000 Sigma: 0.005590 Min: 0.013 Max: 0.027 Range: 0.015
Recv: 13 of 20: z: 0.030 Mean: 0.020769 Sigma: 0.005996 Min: 0.013 Max: 0.030 Range: 0.017
Recv: 14 of 20: z: 0.030 Mean: 0.021429 Sigma: 0.006247 Min: 0.013 Max: 0.030 Range: 0.017
Recv: 15 of 20: z: 0.027 Mean: 0.021833 Sigma: 0.006223 Min: 0.013 Max: 0.030 Range: 0.017
Recv: 16 of 20: z: 0.030 Mean: 0.022344 Sigma: 0.006341 Min: 0.013 Max: 0.030 Range: 0.017
Recv: 17 of 20: z: 0.030 Mean: 0.022794 Sigma: 0.006410 Min: 0.013 Max: 0.030 Range: 0.017
Recv: 18 of 20: z: 0.030 Mean: 0.023194 Sigma: 0.006444 Min: 0.013 Max: 0.030 Range: 0.017
Recv: 19 of 20: z: 0.030 Mean: 0.023553 Sigma: 0.006454 Min: 0.013 Max: 0.030 Range: 0.017
Recv: 20 of 20: z: 0.035 Mean: 0.024125 Sigma: 0.006767 Min: 0.013 Max: 0.035 Range: 0.022
Recv: Finished!
Recv: Mean: 0.024125 Min: 0.013 Max: 0.035 Range: 0.022
Recv: Standard Deviation: 0.006767
```
Just looking at this with a range of +/- 0.022mm is this also still good or do I have an inoptimal probe?

... with `ADAPTIVE_STEP_SMOOTHING `
```
Send: M48 V4 P20
Recv: M48 Z-Probe Repeatability Test
Recv: Positioning the probe...
Recv: Bed X: 117.00 Y: 109.00 Z: -0.00
Recv: 1 of 20: z: 0.007 Mean: 0.007500 Sigma: 0.000000 Min: 0.007 Max: 0.007 Range: 0.000
Recv: 2 of 20: z: 0.010 Mean: 0.008750 Sigma: 0.001250 Min: 0.007 Max: 0.010 Range: 0.003
Recv: 3 of 20: z: 0.013 Mean: 0.010000 Sigma: 0.002041 Min: 0.007 Max: 0.013 Range: 0.005
Recv: 4 of 20: z: 0.015 Mean: 0.011250 Sigma: 0.002795 Min: 0.007 Max: 0.015 Range: 0.007
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```
well didn't go through hahaha. I tried a second time, and it failed on the 3rd probing
```
Send: M48 V4 P20
Recv: M48 Z-Probe Repeatability Test
Recv: Positioning the probe...
Recv: Bed X: 117.00 Y: 109.00 Z: 0.00
Recv: 1 of 20: z: 0.010 Mean: 0.010000 Sigma: 0.000000 Min: 0.010 Max: 0.010 Range: 0.000
Recv: 2 of 20: z: 0.017 Mean: 0.013750 Sigma: 0.003750 Min: 0.010 Max: 0.017 Range: 0.007
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```

PS: @thisiskeithb what does MCU mean?

thisiskeithb, 2022-04-03:
> what does MCU mean

**M**icro**c**ontroller **u**nit. Also known as the processor.

Dids, 2022-05-03:
I can also confirm that `ADAPTIVE_STEP_SMOOTHING` needs to be disabled for this particular board + BLTouch.
Otherwise the probe will fail to trigger at seemingly random, eg. when running 10 probe tests, the 6th step will often fail in my case.

CRCinAU, 2022-05-03:
@Dids - The Marlin Configurations repo has been updated to reflect this - as such unless you enable it yourself, its likely to be fine.

Dids, 2022-05-03:
> @Dids - The Marlin Configurations repo has been updated to reflect this - as such unless you enable it yourself, its likely to be fine.

Thanks! I wasn't fully aware if it's been applied to the config files or not, however I do see that for example for the Ender 3 V2, it is still unfortunately enabled, likely because it's enabled in the stock configurations and was previously overlooked.

I added links to a few examples below for the Ender 3 V2, showing the example configurations still having `ADAPTIVE_STEP_SMOOTHING` enabled by default.

- https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Creality/Ender-3%20V2/BigTreeTech%20SKR%20Mini%20E3%20v3/CrealityUI/Configuration_adv.h#L1202
- https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Creality/Ender-3%20V2/BigTreeTech%20SKR%20Mini%20E3%20v3/MarlinUI/Configuration_adv.h#L1202

---

Slightly off topic: I've seen various discussions before regarding different BTT boards requiring `ENDSTOP_INTERRUPTS_FEATURE` to be disabled, but what I _haven't_ seen is anything definitive on this. Some are saying it should be disabled, as it'll otherwise interfere with homing and/or probing, but it might only apply to older models, eg. not to the SKR Mini E3 V3, but only to V2 etc.

CRCinAU, 2022-05-03:
@Dids Feel free to submit a PR against import-2.0.x to correct those.

`ENDSTOP_INTERRUPTS_FEATURE` should be enabled on any board that has interrupt capable pins for these features. All of the SKR Mini E3 boards I've touched do have this - so it'd be silly to argue that you should turn it off imho....

thisiskeithb, 2022-05-03:
I just put in https://github.com/MarlinFirmware/Configurations/pull/727 to disable `ADAPTIVE_STEP_SMOOTHING` on Ender-3 V2 configs with an SKR Mini E3 V3 motherboard.

CRCinAU, 2022-05-03:
Thanks @thisiskeithb. I'm a bit under the pump on other things at the moment to sit down and not screw up doing that hahaha

thisiskeithb, 2022-05-03:
I thought about adding a Marlin-level warning, but in reality, enabling too many features on a build will cause the same issues most likely. I just need to pay attention to new builds with the E3 V3 motherboard when they are added to the configs repo.

LazeMSS, 2022-05-06:
@thisiskeithb so basically the V3 board is less powerfull than V2?

CRCinAU, 2022-05-06:
Yes and no. It's just different.

There's better protection on common failure parts, more controllable fan headers, better heatsinking, etc etc etc...

If all you look at is CPU Mhz, then yes - but that's only part of the story... imho, `ADAPTIVE_STEP_SMOOTHING` is not a must have feature anyway...