MarlinFirmware/Marlin #23959: [BUG] When using UBL, G1 sets incorrect Z position

eromaredo, 2022-03-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

It might be just mine configuration but when mesh is calculated, Z Offset set and `G28 | G1 Z0` commands sent, nozzle stops ~1.9 mm above a bed and LCD display on Ender shows `Z: 0`. When I try manually move bed to negative number (which is impossible as [Software Endstops](https://marlinfw.org/docs/gcode/M211.html) are enabled), bed moves closer to the nozzle and status still displays `Z: 0`. Additionally sending `M211 S0 | G1 Z-1.9` commands moves nozzle to the _correct_ position. Z-Offset is set to `-2.4` for me so it _might_ not related.

Such behavior makes printing impossible as print starts ~1.9 above the bed. After disabling UBL with `G29 D` command, `G1 Z0` moves bed correctly to the nozzle. Also the issue doesn't appear when `ABL Bilinear Mesh` is active.

Might be related to https://github.com/MarlinFirmware/Marlin/issues/22322.

### Bug Timeline

Hard to say as I'm new here. 

### Expected behavior

`G1 Z0` moves bed correctly to the nozzle.
Manual Z Axis movement from `0` to `0` doesn't change bed position when UBL is active.

### Actual behavior

`G1 Z0` moves bed to the correct position.
`G1 Z0` and manual Z Axis movement produces same z height when UBL is active.

### Steps to Reproduce

1. Create mesh (I'm using this code) https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md#unified-bed-leveling-ubl
2. `G29 A` // To make sure bed leveling is on
3. M851 Z<offset>
4. `G28`
5. `G1 Z0` // Nozzle will be __incorrectly__ positioned above the bed
6. Bonus step - knob Z movement from `0` to `0` (with Software Endstops enabled), moves Z axis closer to a nozzle
7. `G29 D`
8. `G1 Z0` // Nozzle will be __correctly__ positioned above the bed

### Version of Marlin Firmware

Marlin 2.0.9.3 (commit ad94501)

### Printer model

Ender 5 Pro

### Electronics

Stock electronics

### Bed Leveling

UBL Bilinear mesh

### Host Software

OctoPrint

### Additional information & file uploads

```
Bed Topography Report:

    (  1,219)                                                              (219,219)
        0       1       2       3       4       5       6       7       8       9
 9 | +0.083  +0.063  +0.043  +0.086  +0.083  +0.073  +0.094 [+0.099] +0.104  +0.109
   |
 8 | +0.083  +0.063  +0.043  +0.086  +0.083  +0.073  +0.094  +0.099  +0.104  +0.109
   |
 7 | +0.124  +0.108  +0.091  +0.117  +0.095  +0.086  +0.119  +0.130  +0.142  +0.153
   |
 6 | +0.169  +0.158  +0.147  +0.175  +0.172  +0.134  +0.177  +0.177  +0.177  +0.177
   |
 5 | +0.198  +0.180  +0.163  +0.187  +0.155  +0.143  +0.214  +0.224  +0.234  +0.244
   |
 4 | +0.217  +0.194  +0.172  +0.178  +0.154  +0.124  +0.189  +0.187  +0.187  +0.187
   |
 3 | +0.238  +0.203  +0.168  +0.183  +0.160  +0.125  +0.175  +0.184  +0.193  +0.202
   |
 2 | +0.264  +0.207  +0.149  +0.145  +0.135  +0.086  +0.128  +0.139  +0.150  +0.162
   |
 1 | +0.173  +0.159  +0.145  +0.100  +0.083  +0.051  +0.073  +0.074  +0.075  +0.076
   |
 0 | +0.173  +0.159  +0.145  +0.100  +0.083  +0.051  +0.073  +0.074  +0.075  +0.076
        0       1       2       3       4       5       6       7       8       9
    (  1,  1)                                                              (219,  1)
```
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8358260/Configuration.zip)

iamsharkman, 2022-03-29:
I too am having a similar problem. However, my problem is focused on using Z hop. If I use the Z hop feature, the bed doesn't return to its previous hit. This appears to be somewhat intermittent. However, I am currently experiencing the problem everytime I try to print and I get to a particular layer (66) which is the first time Z hop is used. I am using the latest PrusaSlicer software and do not see any anomalies in the G code. I do not use any kind of BLTouch or any other auto leveling devices. Up until the first Z hop the print is perfect. I am also running the latest Marline firmware, v2.0.9.3, on a BTT SKR 2.0 board with TMC2130 drivers on a Tevo Tornado 3D printer which has scores of upgrades and modifications.

sinole1987, 2022-03-30:
There is definitely an issues with probing an z offset. depending on how much I set offset and how much temperature of prob and bed is, I might get "Error probe failed" message. I couldn't figure it out yet, but I replaced multiple probes on my printer and I still have probing issues on new marlin.

closed 2022-06-10 by github-actions[bot]