MarlinFirmware/Marlin #23963: Wrong Russian translate

MM3DMaker, 2022-03-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Please tft_Language_ru with mine because i made the translation more correct.

### Version of Marlin Firmware

Latest

### Additional information & file uploads

Link for tft_Language_ru.h:https://dl.uploadgram.me/6242dbbeb92dcg?raw

N8boy, 2022-03-29:
Please take care before clicking on the link! VirusTotal marked it in 3 cases as Malicious!

@MM3DMaker It would be better to create pull request if you want your code to be implemented.

ellensp, 2022-03-29:
They also converted mm to MM which I don't agree with 

`The millimetre (international spelling; SI unit symbol mm) or millimeter (American spelling) is a unit of length in the metric system, equal to one thousandth of a metre, which is the SI base unit of length.`

MM3DMaker, 2022-03-29:
> Please take care before clicking on the link! VirusTotal marked it in 3 cases as Malicious!
> 
> @MM3DMaker It would be better to create pull request if you want your code to be implemented.

If you are scared heres the script text

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

//****************俄语***************************//
#define TOOL_TEXT_RU                    "инструмент"
#define PREHEAT_TEXT_RU                 " нагрев"
#define MOVE_TEXT_RU                    "двигать"
#define HOME_TEXT_RU                    "дом"
#define PRINT_TEXT_RU                   "печать"
#define EXTRUDE_TEXT_RU                 "экструзия"
#define LEVELING_TEXT_RU                "уровень"
#define AUTO_LEVELING_TEXT_RU           "aвтоуровень"
#define SET_TEXT_RU                     "настройки"
#define MORE_TEXT_RU                    "больше"
#define MORE_GCODE_RU                   "G-код"
#define MORE_ENTER_GCODE_RU             "Введите G-код"

#define ADD_TEXT_RU                     "добавить"
#define DEC_TEXT_RU                     "уменьшить"
#define EXTRUDER_1_TEXT_RU              "экструдер1"
#define EXTRUDER_2_TEXT_RU              "экструдер2"
#define HEATBED_TEXT_RU                 "стол"
#define TEXT_1C_RU                      "1°Ц"
#define TEXT_5C_RU                      "5°Ц"
#define TEXT_10C_RU                     "10°Ц"
#define CLOSE_TEXT_RU                   "выкл"

#define BACK_TEXT_RU                    "назад"

#define TOOL_PREHEAT_RU                 "нагрев"
#define TOOL_EXTRUDE_RU                 "экструдер"
#define TOOL_MOVE_RU                    "двигать"
#define TOOL_HOME_RU                    "дом"
#define TOOL_LEVELING_RU                "уровень"
#define TOOL_AUTO_LEVELING_RU           "aвтоуровень"
#define TOOL_FILAMENT_RU                "замена"
#define TOOL_MORE_RU                    "больше"

#define AXIS_X_ADD_TEXT_RU              "X +"
#define AXIS_X_DEC_TEXT_RU              "X -"
#define AXIS_Y_ADD_TEXT_RU              "Y +"
#define AXIS_Y_DEC_TEXT_RU              "Y -"
#define AXIS_Z_ADD_TEXT_RU              "Z +"
#define AXIS_Z_DEC_TEXT_RU              "Z -"
[... 273 lines cut ...]
#define Y_TMC_MODE_RU                   "включает ли ось Y режим StealthChop"
#define Z_TMC_MODE_RU                   "включает ли ось Z режим StealthChop"
#define E0_TMC_MODE_RU                  "включает ли E0 режим StealthChop"
#define E1_TMC_MODE_RU                  "включает ли E1 режим StealthChop"

#define ADVANCED_CONF_TITLE_RU          "Hастройки принтера>Pасширенные"
#define PAUSE_POSITION_RU    "Hастройки позиции паузы печати"
#define PAUSE_POSITION_X_RU  "положение по X (абс. полож., -1 недействит.)"
#define PAUSE_POSITION_Y_RU  "положение по Y (абс. полож., -1 недействит.)"
#define PAUSE_POSITION_Z_RU  "положение по Z (абс. полож., -1 недействит.)"

#define OFFSET_TITLE_RU  "Hастройки принтера>отступ"
#define OFFSET_X_RU      "X отступ"
#define OFFSET_Y_RU      "Y отступ"
#define OFFSET_Z_RU      "Z отступ"

#define FILAMENT_CONF_TITLE_RU          "Hастройки принтера>Hастройки филамента"
#define FILAMENT_IN_LENGTH_RU           "длинна загрузки"
#define FILAMENT_IN_SPEED_RU            "скорость загрузки"
#define FILAMENT_TEMPERATURE_RU         "температура филамента"
#define FILAMENT_OUT_LENGTH_RU          "длинна извлечения"
#define FILAMENT_OUT_SPEED_RU           "скорость извлечения"

#define ENCODER_CONF_TITLE_RU           "Hастройки принтера>Hастройки энкодера"
#define ENCODER_CONF_TEXT_RU            "энкодер используется?"

thisiskeithb, 2022-03-29:
> Please take care before clicking on the link! VirusTotal marked it in 3 cases as Malicious!

Here you go: [tft_Language_ru.h.zip](https://github.com/MarlinFirmware/Marlin/files/8370771/tft_Language_ru.h.zip)

karliss, 2022-03-29:
> They also converted mm to MM which I don't agree with

Those are not capital latin M but lower case Cyrillic м . Russian language has set of unit symbols that match with the spelling of full unit names. For example kg and Hz, can be written as кг and Гц. Whichever style is chosen it should probably be applied consistently for all units, including mA in X_TMC_CURRENT. I hadn't seen "°Ц" being used before including texts that use Russian spelling for rest of the units, but I am not a native speaker so I can't guarantee that it's never used. All the documents linked by Russian language version of Wikipedia article about SI units that I looked at, used either "C" (capital latin C) or "С" (Cyrillic capital letter Es) even in the column for Russian spelling of units.

MM3DMaker, 2022-03-29:
> Those are not capital latin M but lower case Cyrillic м . Russian language has set of unit symbols that match with the spelling of full unit names. For example kg and Hz, can be written as кг and [...]

°Ц=° means degrees and Ц means Цельсия which in English means Celcius so its also correct.

alexxy, 2022-03-29:
I didnt seen that anywhere in literature symbol
```
°Ц
```
was used for Celcius.

I can create pull for this changes with minor corrections if needed (since i dont agree with some translations)

MM3DMaker, 2022-03-30:
> I didnt seen that anywhere in literature symbol
> 
> ```
> °Ц
> ```
> 
> was used for Celcius.
> 
> I can create pull for this changes with minor corrections if needed (since i dont agree with some translations)

Как хотите

MM3DMaker, 2022-03-30:
The problem is that the original translation was translated by google translate and many of the words do not make sense.

ellensp, 2022-04-01:
I have created a PR from the changed provided. Closing this issue

closed 2022-04-01 by ellensp