MarlinFirmware/Marlin #23988: [BUG] Status text not shown on Ender 3 V2 DWIN MarlinUI

The-EG, 2022-04-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The status text is no longer shown on the status screen when using a genuine 'DWIN' screen shipped with the Ender 3 v2 using MarlinUI. When the status changes, it flashes briefly and then disappears. The status screen also flickers a bit.

### Bug Timeline

49539463b8ed8b15286014b122f87962b7c0f6f4 from #23949

### Expected behavior

The status text to be shown properly

### Actual behavior

The status text is no longer shown on the status screen when using a genuine 'DWIN' screen shipped with the Ender 3 v2 using MarlinUI. When the status changes, it flashes briefly and then disappears. The status screen also flickers a bit.

Reverting 49539463b8ed8b15286014b122f87962b7c0f6f4 results in expected behavior and no flickering.

### Steps to Reproduce

1. Use appropriate MarlinUI config on an ender 3 v2 with a genuine DWIN LCD with encoder.
2. Flash
3. Boot up
4. Observe status screen
5. (Optional) Send additional status messages, note not shown on LCD

### Version of Marlin Firmware

Bugfix-2.0.x as of 2022-04-02 (d98c61cafba1df39316f90af21b99d6c7183f58c)

### Printer model

Ender 3 v2

### Electronics

4.2.2 board, DWIN LCD

### Bed Leveling

UBL Bilinear mesh

### Additional information & file uploads

[ender3v2.zip](https://github.com/MarlinFirmware/Marlin/files/8402531/ender3v2.zip)

tome9111991, 2022-04-02:
I have same issue.
Always refreshing/blinking if i'm scrolling.
https://youtube.com/shorts/iNm_I5J0zCc?feature=share

I use skr e3 turbo board 
With latest bugfix verson from now.

thisiskeithb, 2022-04-03:
@InsanityAutomation: Do you have an (original) DWIN screen and can you take a look at this? Your Ender-3 S1 PR seems have broken some things https://github.com/MarlinFirmware/Marlin/pull/23949

thisiskeithb, 2022-04-05:
> Always refreshing/blinking text if i'm scrolling.

I pinged InsanityAutomation on Discord and he confirmed the UI/screen flashing bug, but couldn't reproduce the status missing text screen bug.

Do these screens have revisions printed on them by chance? Perhaps it's a difference of hardware/DWIN features.

thisiskeithb, 2022-04-06:
@The-EG: Can you upload a video showing the status screen bug?

> Do these screens have revisions printed on them by chance?

To answer my own question, I found my E3V2 DWIN screen and couldn’t find any real markings to indicate board revisions. I’ve been busy IRL, but still plan to test this on my end.

tome9111991, 2022-04-06:
If i insert a SD card  i see the message "SD insert" for 1 sec then is status text clean.

Before the bug
I insert a SD card and i can the the message "SD insert" all the time.

The-EG, 2022-04-07:
_Current bugfix behavior_:
https://youtu.be/mOol4y_RkEc
[image]

It's worth pointing out: either I didn't notice how bad this was before or it's gotten worse in the past few days. Not only is the status text missing, some other parts of the status screen are too. Even the images shown during bootup are different.

_After reverting 49539463b8ed8b15286014b122f87962b7c0f6f4_:
https://youtu.be/O7h0h0sIxSE
[image]

tititopher68-dev, 2022-04-07:
To fix in https://github.com/MarlinFirmware/Marlin/commit/49539463b8ed8b15286014b122f87962b7c0f6f4
Remove all `DWIN_UpdateLCD();` in PR.

thisiskeithb, 2022-04-07:
> To fix in [4953946](https://github.com/MarlinFirmware/Marlin/commit/49539463b8ed8b15286014b122f87962b7c0f6f4) Remove all `DWIN_UpdateLCD();` in PR.

I assume that'll break non-DWIN screens. @InsanityAutomation can you test this fix on your end?

InsanityAutomation, 2022-04-07:
I'll probably have some more tweaks this weekend. They're not all required, some were insurance. Looks like the above may be triggered if status scrolling is disabled, so enabling that is a workaround for the moment.

thisiskeithb, 2022-04-09:
This should be fixed in https://github.com/MarlinFirmware/Marlin/pull/24013

closed 2022-04-09 by thisiskeithb

The-EG, 2022-04-23:
Sorry for the delay, #24013 does fix this issue.

tome9111991, 2022-04-29:
i can confirm fix works good.

only have in some menus kind of color overlay.
[image]