MarlinFirmware/Marlin #23996: [FR] MPC Tuning - Menu

ETE-Design, 2022-04-05:
### Describe the feature you want

Now that MPC Tuning is introduced, it would be great to have an Menu on LCD just like PID Tuning.

thisiskeithb, 2022-04-05:
See https://github.com/MarlinFirmware/Marlin/pull/23984

ETE-Design, 2022-04-05:
@thisiskeithb But #23984 dosen't involve a LCD Menu like PID does? You have to send external command, or make a custom command, would it be better to have it under "Temperature" like PID is?

GMagician, 2022-04-05:
@ETE-Design, please note that PID may be still present because of Bed or chamber

tombrazier, 2022-04-05:
> would it be better to have it under "Temperature" like PID is?

That's where I would be inclined to put it. It's certainly where I would look for it as a user.

> PID may be still present because of Bed or chamber

So both PID and MPC will be in one menu but that seems okay to me. That menu already gets a whole block of options for every heater. It's mostly just a matter of the options being MPC ones rather than PID ones.

tombrazier, 2022-04-06:
Just added this to #23984.

closed 2022-04-06 by thisiskeithb