MarlinFirmware/Marlin #23997: [FR] Add a new flag in Configuration_adv.h to enable auto reporting the reading of a redundant tempurature sensor

StreckerCM, 2022-04-05:
### Is your feature request related to a problem? Please describe.

Marlin has the framework for this available but reporting the redundant sensor is hardcoded to false

### Are you looking for hardware support?

N/A

### Describe the feature you want

In temperature.h

Add a check for the auto-report redundant flag. If it is enabled, print_heater_states would have the include_r flag set to true if it is not, it would have it set to false.

> `#if HAS_TEMP_SENSOR
> 
>     #if  ENABLED(AUTO_REPORT_REDUNDANT)
>       static void print_heater_states(const uint8_t target_extruder
>       OPTARG(HAS_TEMP_REDUNDANT, const bool include_r=true)
>       );
>     #else
>       static void print_heater_states(const uint8_t target_extruder
>       OPTARG(HAS_TEMP_REDUNDANT, const bool include_r=false)
>       );
>     #endif
>     #if ENABLED(AUTO_REPORT_TEMPERATURES)
>       struct AutoReportTemp { static void report(); };
>       static AutoReporter<AutoReportTemp> auto_reporter;
>     #endif
>   #endif` 

In Configuration_adv.h

Add a condition that if AUTO_REPORT_TEMPERATURES is enabled and TEMP_SENSOR_REDUNDANT is defined. If those conditions are met, then AUTO_REPORT_REDUNDANT can be enabled. By default, AUTO_REPORT_REDUNDANT would be commented out and would be an option to be enabled when configuring Marlin

> `/**
>  * Auto-report temperatures with M155 S<seconds>
>  */
> #define AUTO_REPORT_TEMPERATURES
> 
> #if TEMP_SENSOR_REDUNDANT && ENABLED(AUTO_REPORT_TEMPERATURES)
>   //#define AUTO_REPORT_REDUNDANT  //Enabled if you are using a redundant temperature sensor and want to auto report the value
> #endif`

thisiskeithb, 2022-04-06:
Have you tried your patch? If it works, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

thinkyhead, 2022-04-10:
This was simple enough to implement on the fly. See the PR linked above.

closed 2022-04-10 by thinkyhead

StreckerCM, 2022-04-10:
@thinkyhead Thank you!  Your solution is a little more elegant than mine. 
@thisiskeithb I read the contribution page but had not gotten the time to make the patch before @thinkyhead did.

I appreciate the help!