MarlinFirmware/Marlin #24003: [FR] Custom deploy and stow g-code routine for sled microswitch probes.

radry, 2022-04-06:
### Is your feature request related to a problem? Please describe.

Marlin should have (better) support for Microswitch deployable Probes like [QuickDraw ](https://github.com/Annex-Engineering/Quickdraw_Probe)or [Klicky-Probe.](https://github.com/jlas1/Klicky-Probe)

Currently Marlin support "sled" Probes but can only pick them up from the end of the x-axis.
Thus the deploy and stow procedure is currently very limited and not easily configurable: It can only go to the end of the X-Axis + an offset.

Even the currently implemented "TOUCH_MI_PROBE" is too unflexible as it also can only go to the end of the x-axis.

### Describe the feature you want

An easy way (for example in configuration.h) to specify a custom g-code routine how to "deploy" and "stow" the probe.

radry, 2022-04-10:
It seems like Z_PROBE_ALLEN_KEY can be (ab)used for such probes. Although it seems to have some problems as dicussed in issue [#22794](https://github.com/MarlinFirmware/Marlin/issues/22794)

nionio6915, 2022-04-28:
I am the guy behind Euclid probe. I have been wrestling with this for 2+  years.... the issue transcends the pickup routines. 
Probe pickup ad dropoff is not consistent depending on what you are doing with it- probe a single point, do a bed mesh, bed tramming, 401 / 402. All execute inconssitently.
  
You dont see it with tnings like BLTouch becasue the 'deploy' is a short servo motion to move a pin. 

This is not a feature request, this is bug, or number of bugs within what I think is un-coordinated code.

radry, 2022-04-28:
> auto.g approach is one way, and to have gcodes for M401 and M402 defined. then a LOT of the goofiness will go away. problem is, not eveyone has an SD card installed and there is no easy way to emulated.

How would auto.g solve this? The probe needs to be accessible all the time, not only during startup.
Defining M401, M402 should be easy in Marlin but the different probe functions don't use those commands but custom routines.

I'm no expert in Marlin code but having a quick look at the code, there are functions for deploying and stowing a probe. When using "allen key" type, it's possible to define 5 (or so) gcodes lines that are supposed to be executed to deploy and stow. So on first glance it all looks clean. It's just not very user friendly. There is no indication why it would act goofy.

nionio6915, 2022-06-17:
The custome ruotines are exactly the issue. Their behavior is no consistent or predicatble.  

Its not the allen key deployment that is the issue. Marlin does not behave the same pre- and post- probe deploy and stow in all cases. 

If the machine is instructed to deploy for mesh bed leveling it goes through one set of its own motions seemingly in advance of deploy. Then it returns to another XYZ point upon stow.  

If it is instructed to deploy as part of the bed tramming operations. its completely different motions pre-deploy. Stowing then puts the probe back and the carriage goes to in another XYZ place entirely.  iirc, its also different via LCD and command line too. 

If you try to issue a single G30, the machine will go and get the probe, probe, then stow it. 
If you deploy it via M401, then do a G30 at some X Y, it will probe then stow it instead of keeping it deployed.

radry, 2022-06-18:
@nionio6915 

But why is this the case? I had a look at the code in proce.cpp. If allen key is enabled it should always call "run_deploy_moves_script()" and "run_stow_moves_script()"

The different behaviour you experience has nothing to do with the probe deploy and probe stow routine. 
For example G30 is defined in G30.cpp and at the end it says: 
 ` if (raise_after == PROBE_PT_STOW)
    probe.move_z_after_probing();`

So stowing after probing in this case is not a bug but intended. If it's good behaviour or not is open for debate.

As for UBL mesh: I can only see that G29 P1 calls probe.deploy() which in turn (in probe.cpp) calls run_deploy_moves_script() again. I can't find anything that would cause additional moves, unless it was sepecified somewhere else in the configuration, regardless of probe type. Which means this "erratic" movement would happen even with other probes.

nionio6915, 2022-06-19:
the the bottom line is that the probe deploy and stow is hard coded into the firmware and gives the user few options for customization. the deploy and stow behavior IS different between use cases  G29, G30, G34 nd G35.

thisiskeithb, 2022-07-07:
https://github.com/MarlinFirmware/Marlin/pull/24420 has been merged.