MarlinFirmware/Marlin #24005: Invalid function parameter

doktorzaba, 2022-04-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

an error is reported when calling the print_2d_array function in abl.cpp (row 139): 
"Marlin\src\feature\bedlevel\abl\abl.cpp:141:3: error: cannot convert 'print_bilinear_leveling_grid()::<lambda(uint8_t, uint8_t)>' to 'const float*"

### Version of Marlin Firmware

2.0.9.3

### Bed Leveling

ABL Bilinear mesh

### Host Software

OctoPrint

ellensp, 2022-04-07:
No configs, not steps to reproduce. 
Please fill out the template with requested information

doktorzaba, 2022-04-07:
Thanks, I'm adding config.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/8439472/Configs.zip)

ellensp, 2022-04-07:
Provided configuration compiles without issues

But they will not works as is

This makes zero sense on a octopus
```
  #define X_MAX_PIN  3 // creality connects X_MAX switch to X_MIN(_PIN) motherbard connector
  #define Y_MAX_PIN 14 // creality connects Y_MAX switch to Y_MIN(_PIN) motherbard connector

```

doktorzaba, 2022-04-07:
I'm upgrading an existing configuration and I'm not done yet, so it contains invalid lines. However, this is not related to this error. The error occurred after enabling #define AUTO_BED_LEVELING_BILINEA
[image]
R

ellensp, 2022-04-07:
Something wrong with your copy of marlin, as provided configs compiles without issues (with lots of warning due to redefined endstop pins) 

[image]

doktorzaba, 2022-04-07:
So everything is deleted, reinstalled. Downloaded new clean Marlin, including with bug fixes https://marlinfw.org/meta/download/
I only allowed these two values
AUTO_BED_LEVELING_BILINEAR
PROBE_MANUALLY
Otherwise everything remained the same, compilation for mega2560
result:
[image]

ellensp, 2022-04-07:
Attach full build log. as same to for me compiles without issues

doktorzaba, 2022-04-07:
I'm sending output from the environment after compilation, maybe that's what you need :)
[logs.zip](https://github.com/MarlinFirmware/Marlin/files/8442649/logs.zip)

ellensp, 2022-04-07:
this makes zero sense...  Im using same versions.... without issues

doktorzaba, 2022-04-07:
When I comment function call print_2d_array, the compilation passes.

thisiskeithb, 2022-04-07:
> Something wrong with your copy of marlin, or platformio as provided configs compiles without issues (with lots of warning due to redefined endstop pins)

I'm seeing the same thing here. Other than the wall of warnings from endstop redefines, it compiles fine.

closed 2022-04-07 by thisiskeithb

ellensp, 2022-04-08:
Where exactly did you download code from. Ie the full url
How did you download the code, zip file? Git clone?  other?

doktorzaba, 2022-04-08:
I have already downloaded several times from this address https://marlinfw.org/meta/download/ including bugfix. I will delete all previous attempts, create a new directory, copy the files, leave everything to the default, just enable AUTO_BED_LEVELING_BILINEAR +
PROBE_MANUALLY and run the compilation for mega 2560. Already tested on 2 PCs, with a clean installation of VS code and still the same.

doktorzaba, 2022-04-08:
This is my last installation
[Marlin-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/8448867/Marlin-2.0.x.zip)

ellensp, 2022-04-08:
You have extracted bugfix over 2.0.x
So you directory has some old and some new code
Marlin/src/feature/bedlevel/abl/abl.cpp should not be in bugfix code
Marlin/src/feature/bedlevel/abl/abl.h should not be in bugfix code

see https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/feature/bedlevel/abl

doktorzaba, 2022-04-08:
So sorry, it was my fault. I first extracted 2.0.x.zip and then bugfix-2.0.x.zip. If I understand correctly, only bugfix-2.0.x.zip should be extracted, then the compilation will work fine.
Thank you very much for your help.