MarlinFirmware/Marlin #24007: [BUG] Compile error when enabling BLTOUCH ( Servo related) on ESP32 (MKS Tinybee)

jesterhead82, 2022-04-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Compile error on latest Bugfix when enabling BLtouch.

Compiling .pio\build\mks_tinybee\src\src\MarlinCore.cpp.o
Marlin/src/HAL/ESP32/Servo.cpp: In member function 'int8_t Servo::attach(int)':
Marlin/src/HAL/ESP32/Servo.cpp:38:42: error: 'get_pwm_channel' was not declared in this scope
   channel = get_pwm_channel(pin, 50u, 16u);
                                          ^
Compiling .pio\build\mks_tinybee\src\src\core\serial.cpp.o
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mks_tinybee\src\src\HAL\ESP32\Servo.cpp.o] Error 1

### Bug Timeline

At keast as of April/2022

### Expected behavior

Compilation when BLtouch is enabled on ESP32 (MKS Tinybee)

### Actual behavior

Comnpile failure

### Version of Marlin Firmware

2.0.xx

### Additional information & file uploads

[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/8441005/MarlinConfig.zip)

ellensp, 2022-04-09:
Unrelated: provided configs have an error
`#define BABYSTEP_INVERT_Z trur           // Change if Z babysteps should go the other way`

jesterhead82, 2022-04-10:
Yep ;)

Got it compiling by adding get_pwm_channel to MarlinHAL, had to also bypass issue from  #24021, but was not able to test it yet

jesterhead82, 2022-04-20:
Hi,

tried again with todays bugfix, enabling Servo (e.g. simply by enabling BLTOUCH probe) still does not compile for me, so workaround for me was to add  get_pwm_channel in class MarlinHAL

thisiskeithb, 2022-04-20:
Sorry. I added confirmed labels and didn’t comment the first time, but I can confirm the that I get the same error messages when compiling.

thinkyhead, 2022-04-21:
Now patched. Thanks for the report!

commit d507ea11 (2022-04-21): 🩹 Fix ESP32 servos, SD_IGNORE_AT_STARTUP

Fixes #24007
Followup to aaf5bf02
commit 7444933d (2022-05-07): 🩹 Fix ESP32 servos, SD_IGNORE_AT_STARTUP

Fixes #24007
Followup to aaf5bf02
closed 2022-04-21 by thinkyhead