MarlinFirmware/Marlin #24016: [BUG] MKS Lvl Display + ADVANCED_PAUSE_FEATURE + FILAMENT_RUNOUT_SENSOR = "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."

Liriel-666, 2022-04-10:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

When settin in Marlin 2.0.9.3 and earlier MKS LVL +   ADVANCED_PAUSE_FEATURE + FILAMENT_RUNOUT_SENSOR it can not compiled because error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."      

### Bug Timeline

first for me first appear in 2.0.9.3

### Expected behavior

It should run because Lvl has support of it

### Actual behavior

when compile that came

> Executing task: C:\Users\Laen\.platformio\penv\Scripts\platformio.exe run <

Processing mks_robin_nano_v3 (platform: ststm32@~12.1; board: marlin_STM32F407VGT6_CCM; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Copying MKS Assets
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32F407VGT6_CCM.html
PLATFORM: ST STM32 (12.1.1) > STM32F407VG (128k RAM, 64k CCM RAM, 1024k Flash
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <lvgl> 6.1.1
|-- <qr_encode> 0.0.0+20201117144502
|-- <TMCStepper> 0.7.3
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\Print.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\VirtIOSerial.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\WString.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\abi.cpp.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\board.c.o
Compiling .pio\build\mks_robin_nano_v3\FrameworkArduino\hooks.c.o
[... 244 lines cut ...]
Anet A8

### Electronics

MKS Robin Nano V3 with mks touchscreen and filament runout sensor

### Add-ons

2 extruder + 2 Filament Runout sensor

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8459523/Marlin.zip)

thisiskeithb, 2022-04-10:
If you disable/comment out the sanity check, does it work?

Liriel-666, 2022-04-11:
> 

Where is that sanity check to disable?

What I want is the filament runout sensor to activate.

Emergency Parser? Is it on or off by default because I dont change that

thisiskeithb, 2022-04-11:
> Emergency Parser? Is it on or off by default because I dont change that

It's disabled by default, but you can find it in `Configuration_adv.h`: https://github.com/MarlinFirmware/Marlin/blob/80d09044f7838ae11c97b03f465bd92558fc2c24/Marlin/Configuration_adv.h#L2348-L2356

Enable it and you will not get a compile error.

closed 2022-04-11 by thisiskeithb

Liriel-666, 2022-04-11:
mmh ok emergency parser  was the riddle but now new errors