MarlinFirmware/Marlin #24020: A20T crash and freeze during print

Sebxtreme, 2022-04-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Geeetech A20T a4988 soldered on board Marlin freeze any time during print , latest vertabreak firmware 

### Actual behavior

Freeze during print 

### Version of Marlin Firmware

Vertabreak 

### Printer model

Geeetech A20T

### Electronics

Stock

### Add-ons

None 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Additional information & file uploads

Need heelllppppo

thisiskeithb, 2022-04-10:
> latest vertabreak firmware

We can’t support forks of Marlin, so you’ll need to open an issue in Vertabreak’s fork.

closed 2022-04-10 by thisiskeithb

Sebxtreme, 2022-04-10:
Oh no then I will try the latest original marlin.

Mary S.

Closed #24020<https://github.com/MarlinFirmware/Marlin/issues/24020>.