MarlinFirmware/Marlin #24021: MKS_LCD12864B and MKS_MINI_12864 Linking error on ESP32 (MKS Tinybee)

jesterhead82, 2022-04-10:
Same happens still with todays (2022/04/10) 2.0X-Bugfix, dos compile e.g. with 
MKS_MINI_12864_V3
but not with either 
#define MKS_LCD12864B    (<Triggers also MKS_MINI_12864)
#define MKS_MINI_12864

Linking error:
.pio/build/mks_tinybee/src/src/lcd/dogm/u8g_dev_uc1701_mini12864_HAL.cpp.o:(.data.u8g_dev_uc1701_mini12864_HAL_2x_hw_spi+0x8): undefined reference to `u8g_eps_hw_spi_fn(_u8g_t*, unsigned char, unsigned char, void*)'

I dont know so much of this stuff, but:

ESP32 u8g_esp32_spi.cpp:

#include "../../inc/MarlinConfigPre.h"

#if ENABLED(FYSETC_MINI_12864_2_1)    <--- is this correct?
#include <U8glib-HAL.h>
#include "Arduino.h"

_Originally posted by @jesterhead82 in https://github.com/MarlinFirmware/Marlin/issues/23776#issuecomment-1094376547_

ellensp, 2022-04-11:
MKS marlin fork has  
`#if ANY(MKS_MINI_12864, MKS_MINI_12864_V3)`

Marlin has 
`#if ENABLED(FYSETC_MINI_12864_2_1)`

As the MKS_MINI_12864_V3 has an  identical display the BTT_MINI_12864_V1 so both should work.  Ie
```
// MKS_MINI_12864_V3 and BTT_MINI_12864_V1 are identical to FYSETC_MINI_12864_2_1
#if EITHER(MKS_MINI_12864_V3, BTT_MINI_12864_V1)
  #define FYSETC_MINI_12864_2_1
#endif

```

So the code in marlin should be 

`#if ANY(MKS_MINI_12864, FYSETC_MINI_12864_2_1)`

thisiskeithb, 2022-04-11:
> So the code in marlin should be
> #if ANY(MKS_MINI_12864, FYSETC_MINI_12864_2_1)

The standard MKS Mini 12864 is very different than the newer “mini” 12864s and does not use neopixels as a backlight, so this will need some checking.

thinkyhead, 2022-04-20:
It compiles fine for me.

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

jesterhead82, 2022-04-20:
Hello,

does only compile with todays bugfix for a MKS_MINI_12864 or MKS_LCD12864B if I apply proposed change from ellensp

So in

u8g_esp32_spi.cpp:

from 
#if ENABLED(FYSETC_MINI_12864_2_1)

to
#if ANY(MKS_MINI_12864, FYSETC_MINI_12864_2_1)

thisiskeithb, 2022-04-21:
The regular MKS Mini (V1 and V2) does not use neopixels, so grouping it with the V3 (FYSETC_MINI_12864_2_1 in our case) will enable options it doesn’t have.

ellensp, 2022-04-23:
@thisiskeithb this is a global enable/disable for all code in  Marlin/src/HAL/ESP32/u8g_esp32_spi.cpp 

Ie the SPI communications side only.

It has nothing to do with if it has neopixels or not

ellensp, 2022-04-23:
Pr created.

closed 2022-04-23 by ellensp